dw-0/kiauh #271: kiauh_macros.cfg' does not exist

soulrazr, 2022-12-06:
### Linux Distribution

WSL

### What happened

I installed Klipper, Moonraker, then Fluidd. I compiled and flashed the firmware and when I try to open Fluidd I get this error:
Include file '/home/username/printer_data/config/kiauh_macros.cfg' does not exist
I checked the directory and the file is definitely missing.

### What did you expect to happen

I spent over an hour searching google and going through multiple tutorials and as far as I can tell that is something that is supposed to have been generated when I installed Fluidd.

### How to reproduce

Install Fluidd and select yes to Add the recommended macros

### Additional information

I have no idea what I'm doing with all of this. I've never touched linux or klipper before today and I can't find any answer on what that file is supposed to actually contain.

th33xitus, 2022-12-07:
A file calles "kiauh_macros.cfg" is not used anymore for quite a long time. I used a file like that in the past. So you might have that include statement still in your config. The fluidd installer now pulls the macro file directly from the FluiddPi repository, so i don't have any work keeping their macros up2date, instead i just download them for the user if he request to do so.

soulrazr, 2022-12-07:
Thanks. That did the trick

closed 2022-12-07 by soulrazr