dw-0/kiauh #273: How to handle pipes in gcode_shell_command?

DrKlipper, 2022-12-07:
### Linux Distribution

Raspberry PI OS

### What happened

I would like to execute a command with pipes like this one: 
```
[gcode_shell_command ListPorts]
command: ls -lR /dev/ | grep serial
timeout: 5.
verbose: True
```

But everything after the pipe is ignored or results in errors. 

### What did you expect to happen

Would be nice to know how pipes are working. 

### How to reproduce

Just use the above example

th33xitus, 2022-12-08:
I suggest to use a separate shell script to call that command and inside of that gcode_shell_command you execute that shell script.

manu7irl, 2022-12-13:
@DrKlipper I think the easy part is running the script, but from there, you need to pipe it to console,
like:
`ls -lR /dev/|grep serial > /tmp/printer`
or if you have the new structure folders:
`my_mcu_id=$('ls -lR /dev/|grep serial')  
echo "M118 $my_mcu_id"  > ~/printer_data/comms/klippy.serial`
this shloud pipe that as a command on the console and display it if you have the [respond] section in your config
try it and report.
thanks.

th33xitus, 2022-12-16:
> But since this is only a oneliner the overhead is alot.

@DrKlipper that is true in that case, but i'm not aware of another solution. I didn't write the gcode_shell_command module, but i know that it doesn't work with things like pipes and expanding of e.g. wildcards . That's why i usually always suggest to use an extra shell script for operations that require such stuff. I wouldn't consider it as a bug, rather like intended or like a design decision by the creator of the module.

@manu7irl if verbose is set to true, you get the full terminal output in the console, i assume he is already using mainsail or fluidd, so the output should appear in the interfaces console.

manu7irl, 2022-12-16:
@th33xitus wow didn't know about the verbose thingy...
I need to try it myself!
But isn't it that verbose will fully display the whole stdout and not only what's needed to be shown?

NonaSuomy, 2023-02-18:
Thanks for the tip working nicely!
sudo apt install zbar-tools
```bash
cd /home/pi/klipper/klippy/extras/
wget https://raw.githubusercontent.com/th33xitus/kiauh/master/resources/gcode_shell_command.py
```
```yaml
[gcode_shell_command qrcode_snapshot]
command: wget http://IP/webcam/?action=snapshot -O /home/pi/printer_data/gcodes/qrcode.jpg
verbose: True
timeout: 2.
[gcode_shell_command qrcode_scanner]
command: zbarimg /home/pi/printer_data/gcodes/qrcode.jpg
verbose: True
timeout: 2.
[gcode_macro QRCode_Scan]
gcode:
    RUN_SHELL_COMMAND CMD=qrcode_snapshot
    RUN_SHELL_COMMAND CMD=qrcode_scanner
```
[image]

NonaSuomy, 2023-07-09:
How would I then take 
```
[gcode_shell_command qrcode_snapshot]
command: wget http://IP/webcam/?action=snapshot -O /home/pi/printer_data/gcodes/qrcode.jpg
verbose: True
timeout: 2.
[gcode_shell_command qrcode_scanner]
command: zbarimg /home/pi/printer_data/gcodes/qrcode.jpg
verbose: True
timeout: 2.
[gcode_macro QRCode_Scan]
gcode:
    RUN_SHELL_COMMAND CMD=qrcode_snapshot
    RUN_SHELL_COMMAND CMD=qrcode_scanner
    SET_ACTIVE_SPOOL ID= (Just the serial number from the output of qrcode_scanner) <== What would I do here
``` 

Thank you

closed 2023-10-31 by dw-0