dw-0/kiauh #275: Can FIRMWARE.BIN have the date appended when flashing to Creality boards?

vampyrex13, 2022-12-20:
### Is your feature request related to a problem? Please describe

When flashing to a Creality board, the file name is always FIRMWARE.BIN. If the last flashed file has the same name as the updated file, it will not flash again without having to manually remove the SD card, load it on a computer, and renaming the file.

### Describe the solution you'd like

Update the flashing script to append the date to the end of FIRMWARE.BIN (e.g. FIRMWARE20201220.BIN for December 20th, 2022). This is what I have been manually doing every other firmware flash.

jbhome, 2023-06-08:
Please add the time as well as you may need to do more than one build/flash to get it right

togeis, 2023-06-25:
Need this too. Wondered why new firmware has not been loaded...