dw-0/kiauh #281: Fluidd Download URL Parsing Failure

LongDono, 2023-01-06:
### Linux Distribution

Ubuntu Server 22.04.1 LTS

### What happened

The Fluidd download URL parser is failing, preventing installation of Fluidd. I suspect it is because the API limit has been exceeded. I looked through the code and found the URL it was using to try to identify releases and entered it manually:

{"message":"API rate limit exceeded for 38.20.250.13. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)","documentation_url":"https://docs.github.com/rest/overview/resources-in-the-rest-api#rate-limiting"}

### What did you expect to happen

Download URL to be parsed correctly or if not solvable, a message indicating what the issue is

### How to reproduce

Usual installation steps

### Additional information

###### Downloading Fluidd from https://github.com/fluidd-core/fluidd/releases/download//fluidd.zip ...
--2023-01-05 19:45:25--  https://github.com/fluidd-core/fluidd/releases/download//fluidd.zip
Resolving github.com (github.com)... 140.82.112.4
Connecting to github.com (github.com)|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-01-05 19:45:26 ERROR 404: Not Found.

#=======================================================#
 Downloading Fluidd from
 https://github.com/fluidd-core/fluidd/releases/download//fluidd.zip
 failed!
#=======================================================#

wilsongis, 2023-02-05:
I added this to both download functions for mainsail and fluid. It looks like the scripts don't always get the tag for fluidd or mainsail. I am not exactly sure why. I updated my scripts to download a version (current as of today) in cases where it can't find the tag. I rather have something rather than nothing. It is a hack job. 

```
function download_fluidd() {
  local url
  local badurl
  badurl="https://github.com/fluidd-core/fluidd/releases/download//fluidd.zip"
  url=$(get_fluidd_download_url)
  if [ $url==$badurl ]
  then
    url="https://github.com/fluidd-core/fluidd/releases/download/v1.23.1/fluidd.zip"
  else
    url=$(get_fluidd_download_url)
  fi
  status_msg "Downloading Fluidd from ${url} ..."
```

```
function download_mainsail() {
  local url
  local badurl
  badurl="https://github.com/mainsail-crew/mainsail/releases/download//mainsail.zip"
  url=$(get_mainsail_download_url)
  if [ $badurl == $url ]
  then
    url="https://github.com/mainsail-crew/mainsail/releases/download/v2.4.1/mainsail.zip"
  else
    url=$(get_mainsail_download_url)
  fi
  status_msg "Downloading Mainsail from ${url} ..."
```

Tron842, 2023-03-02:
I had the same issue, and it turned out to be the CA store needing to be updated. You can test if you have the same problem by running the following command.

`curl "https://api.github.com/repos/mainsail-crew/mainsail/releases/latest"`

If you get a  'curl: (60) SSL certificate problem: unable to get local issuer certificate' error, it is probably that. I was able to update my CA store with the following commands:

```
cd ~
curl "https://curl.se/ca/cacert.pem" -k > cacert.pem
sudo mv ./cacert.pem /usr/local/share/ca-certificates/certificate.crt
sudo dpkg-reconfigure ca-certificates
```

Hopefully, this helps.

PR #348 (2023-06-04): fix(mainsail/fluidd): fall back to latest stable url if fetching tags fails
This PR fixes #281
closed 2023-06-04 by th33xitus