dw-0/kiauh #287: Q: Support for init.d

CODeRUS, 2023-01-15:
### Is your feature request related to a problem? Please describe

kiauh currently supports only systemd installation. Manually created intt.d services are being removed by kiauh as unwanted.

### Describe the solution you'd like

Detect if running os is actually init.d based, or just check for systemd existence

### Additional information

This may help to get proper way to install klipper and co. to android/termux, linux4android containers and creality pad (probably even more exists)

I can start working on it and test on android with linux deploy installation of debian. Later can be updated for creality pad if necessary.

th33xitus, 2023-01-19:
Mhhh im not quite sure about that. That would require providing init.d service files for moonraker and literally all other components as well. What im really not interested in is supporting the creality sonic pad in any way. I refuse to support that product at all.

So the general issue or feature you propose is, to not touch any existing init.d klipper services in the klipper remove script? And simply stick to systemd only anymore?

> I can start working on it and test on android with linux deploy installation of debian.

we could think about a special branch maybe... a dedicated branch that supports such operating systems and keep the KIAUH master branch support systemd only? not sure how smart that would be in the long run though.

CODeRUS, 2023-01-19:
> So the general issue or feature you propose is, to not touch any existing init.d klipper services in the klipper remove script? And simply stick to systemd only anymore?

At least yes, the rest of hacks could be done with fake systemctl script

th33xitus, 2023-02-28:
@CODeRUS fyi, once #304 is merged, init.d support is removed: <[`4ebe016` (#304)](https://github.com/th33xitus/kiauh/pull/304/commits/4ebe0162d6fdd80337a0b750c9b3591f8fd77629)> that means, init.d services will also not be removed anymore by kiauh. i think that is what you wanted, right?

CODeRUS, 2023-02-28:
@th33xitus yes that's would be great, thanks!

closed 2023-10-31 by dw-0