dw-0/kiauh #288: Time gcode_shell_command to the surrounding gcodes

m-sundman, 2023-01-22:
### Is your feature request related to a problem? Please describe

Now the `gcode_shell_command` executed way, way before commands that come before it in the gcode file are processed by the mcu. It seems like the shell command is executed after the preceding gcode is sent to the mcu’s input queue, rather than when that gcode is (expected to be) actually processed on the mcu.

### Describe the solution you'd like

Have the shell command execute when that gcode is (expected to be) actually processed on the mcu.

### Describe alternatives you've considered

I've tried to add a delay to the beginning of the shell command, but the time offset is too unpredictable to be anywhere near where (or rather when) it's supposed to be.

closed 2023-03-12 by th33xitus

m-sundman, 2023-03-13:
@th33xitus if gcode_shell_command isn't kiauh (which is where I got it from), then any idea in which project this feature request should be?

th33xitus, 2023-03-13:
@m-sundman I think your issue is rather related to how Klipper evaluates and calls macros, and i think shell command macros are included in that regard. Unless i don't understand your question. In that case i would need to see an issue, but then again its probably also an even better idea to try and get help on the Klipper Discord.