dw-0/kiauh #289: Install Klipper fails every time on line 229

TristanLashmar, 2023-01-24:
### Linux Distribution

Raspian

### What happened

each time I run the script, following the instructions from this git, I get an error when the script calls the copy function at line 229:

`###### Installing single Klipper instance ...

###### Checking for the following dependencies:
● git
[✓ OK] Dependencies already met!

###### Cloning Klipper from https://github.com/Klipper3d/klipper ...
Cloning into '/home/fj/klipper'...
remote: Enumerating objects: 41800, done.
remote: Counting objects: 100% (20/20), done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 41800 (delta 8), reused 8 (delta 5), pack-reused 41780
Receiving objects: 100% (41800/41800), 104.65 MiB | 662.00 KiB/s, done.
Resolving deltas: 100% (32321/32321), done.
/home/fj/kiauh/scripts/klipper.sh: line 229:  1535 Killed                  git clone "${repo}" "${KLIPPER_DIR}"

#=======================================================#
 Cloning Klipper from
 https://github.com/Klipper3d/klipper
 failed!
#=======================================================#
`
This is on a fresh install and I have tried with different SD cards to make sure it wasnt a bad or slow card.

### What did you expect to happen

Klipper install to complete.

### How to reproduce

raspberry Pi 3, clean install of Raspian using the raspberry Pi flash tool, follow steps on this git main page.

### Additional information

Git and packages all updated before running the script. Python version is 3.9.2-3.

th33xitus, 2023-02-05:
Until now i have never seen this issue or was able to reproduce it. My first idea was, that it might be a network issue on your side, so that the clone operation fails due to a failing network connection or something like that. But it looks like everything was fully downloaded. For now im out of ideas why that could be, and im not sure what the root cause of this issue might be. Do you still get this issue? Or did it resolve itself in the meantime?

TristanLashmar, 2023-02-06:
I worked around it by using Mainsail OS. Ive tried this on 3 different Rpi 3B's too, and different cards. Same issue with each. once I got Mainsail OS running the machine would hit 100% CPU constantly and was not usable, I ended up using a thin client + Debian + KIAUH worked perfectly fine on that. It made it a breeze to install - thank you!

I think the Pi3 B is just not capable of doing it any more, despite what Klipper docs say.

th33xitus, 2023-02-06:
A Pi 3B should be possible to use but it really depends on what other components should run on it. For running ONLY Klipper it should be okay, but then there is Input Shaper, Webcams, KlipperScreen, a bot here and there... so yeah it depends. On one printer i run Klipper (only) on a Rpi Zero 2 W, which is (afaik) comparable to the 3B in terms of performance, but there i also see sometimes a timer too close issue.

Anyways...so in conclusion it wasn't actually a KIAUH issue but rather a hardware one? Just so i know how to proceed with this issue (if at all) or can i close it?

TristanLashmar, 2023-02-07:
I think best it is closed yup, hardware. The Pi 3 is really cutting it too fine I think.

Thanks again for making this script, it made life so much easier on the other install.

closed 2023-02-07 by th33xitus