dw-0/kiauh #295: Crowsnest could not be installed

dapengg, 2023-02-12:
### Linux Distribution

Armbian 23.02.0-trunk Bullseye with Linux 6.1.10-flippy-81+

### What happened

[image]
[image]
[image]

### What did you expect to happen

Reinstallation does not take effect and cannot be uninstalled

### How to reproduce

Reinstallation does not take effect and cannot be uninstalled

th33xitus, 2023-02-12:
Seeing all those permission denied messages lead me to the conclusion that something else must be wrong with your system or, obviously, with the current users permissions you run all those steps on. Hence i currently doubt that it's an issue with KIAUH or crowsnest.

MS1987, 2023-02-13:
I met the same issue. And I can confirm that all was well last week. So I think it is relative with the updates of these days. @th33xitus

closed 2023-03-04 by th33xitus