dw-0/kiauh #298: Use fork of KlipperScreen

hyptx, 2023-02-16:
### Is your feature request related to a problem? Please describe

No. 

### Describe the solution you'd like

I am looking for a way to change KlipperScreen to a fork. I know you can switch Klipper to a custom repo, can you switch KlipperScreen, if so how?

th33xitus, 2023-02-16:
That's currently only possible if you modify KIAUHs source code: https://github.com/th33xitus/kiauh/blob/master/scripts/globals.sh#L52
There are no configuration options in place currently.

hyptx, 2023-02-16:
Thanks for the fast reply. I'm elbow deep in KlipperScreen mods and this solution should work great for my purposes.

closed 2023-02-16 by hyptx