dw-0/kiauh #302: Kiauh cannot install crowsnest completly.

MS1987, 2023-02-26:
### Linux Distribution

Armbian with linux 5.16.20

### What happened

On my arm64 + armbian platform, I use the lastest Kiauh to install crowsnest, it seems all the steps are finished, but it always show "incomplete" state. And actually, it cannot be used with my camera. This issue has been existed two weeks, hope there any solution.
[image]

### What did you expect to happen

Hope crowsnest can be installed completly with kiauh.

### How to reproduce

Use the lastest Kiauh to install crowsnest

th33xitus, 2023-02-27:
> This issue has been existed two weeks, hope there any solution.

There is not much more i can say other than what i already said here: https://github.com/th33xitus/kiauh/issues/295#issuecomment-1426981342

skyaee, 2023-03-01:
Kiauh cannot install crowsnest completly

skyaee, 2023-03-01:
It happened to me too, and reinstalling the system didn't fix it.

jsrjrivas, 2023-03-04:
still having this issue, not sure whats going on but the crowsnest service refuses to start. was working fine before updating as well

gnosticJade, 2023-03-10:
I'm unable to install crowsnest through kiauh. This is after installing Klipper & Moonraker. No other errors in the logs prior to these. Full logs on [pastebin](https://pastebin.com/raw/JNXmZDgh).

```
Installing 'crowsnest' Dependencies ... [OK]
Linking crowsnest ... [OK]
Copying crowsnest.conf ... [OK]
Build dependend Stream Apps ...
Cloning ustreamer repository ...
fatal: could not create work tree dir 'bin/ustreamer': Permission denied
make: *** [Makefile:37: install] Error 128
>>>>>> Something went wrong! Please try again...
```

piotrb, 2023-03-18:
Hey folks, I ran into a similar issue and submitted a PR #317 to address the issue I was getting (which manifested the same way as your logs)

Can you confirm if prior to running kiauh, you used `su` or `sudo` to change users?

MS1987, 2023-03-19:
@piotrb Right, I have this issue when I use su to switch non-root user. And I tried to directly use non-root user to login, and it showed the install is complete! But another issue come as I [committed](https://github.com/mainsail-crew/crowsnest/issues/97). I think you found the reason, and I will try your commit.

closed 2023-03-19 by th33xitus