dw-0/kiauh #307: index-pack failed

rafaelldavid, 2023-03-03:
### Linux Distribution

Armbian 23.02.2 Bullseye with Linux 4.4.194-rk322x

### What happened

[image]

### What did you expect to happen

I installed Kiauh and when I select to install Klipper, the program downloads the files but ends up with this error. (PS: Sorry if this is not the right place to post)

### How to reproduce

./kiauh/kiauh.sh

Option 1-1 with python3 and 1 instance of Klipper.

rafaelldavid, 2023-03-03:
[image]

th33xitus, 2023-03-03:
Im sorry but this is not a KIAUH issue or bug, it seems like its some issue with your network connection and the cloning (download) of Klipper failed due to that. Nothing i can do for you. Sorry.

rafaelldavid, 2023-03-03:
Can you tell what could be going on? It is connected to the internet by cable. The internet here is stable...

rafaelldavid, 2023-03-03:
Now a different error appears:

Cloning into '/home/rafael/klipper'...
remote: Enumerating objects: 43729, done.
remote: Total 43729 (delta 0), reused 0 (delta 0), pack-reused 43729
Receiving objects: 100% (43729/43729), 116.31 MiB | 3.26 MiB/s, done.
Resolving deltas: 100% (33901/33901), done.
fatal: did not receive expected object a8687520a3e42acee698e59bc7577d172e72ff98
fatal: index-pack failed

closed 2023-03-12 by th33xitus