dw-0/kiauh #312: After install crowsnest , blink cursor in display

Maurotb, 2023-03-12:
### Linux Distribution

Raspberry pi desktop 11.6

### What happened

I have try 3 times with same result, fresh install in rpi3b with 5'' DSI lcd.
After install crowsnest ,  reboot pi , network, ssh and web gui works, but in display i see only a blinking cursor.
If i remove crowsnest, problem persist 

### What did you expect to happen

Display KlipperScreen in my monitor

### How to reproduce

Install a fresh pi desktop
Install Klipper Fluidd Mainsail,KlipperScreen
Now install crowsnest and reboot

th33xitus, 2023-03-12:
This doesn't look like an issue or a bug with KIAUH since installing everything seemed to work. So it might be related to something else. As im not experienced with KlipperScreen and Crowsnest at all, i can't comment on that.

JNP-1, 2023-03-12:
> ### Linux Distribution
> Raspberry pi desktop 11.6
>
> ### What happened
> I have try 3 times with same result, fresh install in rpi3b with 5'' DSI lcd. After install crowsnest , reboot pi , network, ssh [...]

Has Klipperscreen ever worked after an installation?

Maurotb, 2023-03-12:
> 

Yes work on fresh install. It stop wortking, after install crowsnest. I try 3 times, same problem

JNP-1, 2023-03-12:
> > 
> 
> Yes work on fresh install. It stop wortking, after install crowsnest. I try 3 times, same problem

Can you do a reboot after a new installation (without installing crowsnest )and see if Klipperscreen still works?

Maurotb, 2023-03-12:
Yes. Klipperscreen works after reboot. But now if i install crownsnest , i have a blinking cursor

JNP-1, 2023-03-12:
> Yes. Klipperscreen works after reboot. But now if i install crownsnest , i have a blinking cursor

Very strange. I have never experienced it like this before. However, I always use MainsailOS.