dw-0/kiauh #313: Octoeverywhere not completing install on a fresh Raspbian 11 Image

Endracion, 2023-03-14:
### Linux Distribution

Raspbian 11 Bullseye 32 Bit

### What happened

Tried starting the install for OctoEverywhere, ended in a loop and this error:

Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
Found existing OctoEverywhere service config.
Server section not found in OE config.
The service is taking a while to start, there might be something wrong.
Do you want to keep waiting? [y/n] n

We didn't get a response from the OctoEverywhere service when waiting for the printer id.
You can find service logs which might indicate the error in: /home/pi/printer_data/logs

Attempting to print the service logs:
-- Journal begins at Tue 2023-02-21 01:23:30 GMT, ends at Tue 2023-03-14 16:03:33 GMT. --
Mar 14 16:02:53 raspberrypi systemd[1]: Started OctoEverywhere For Moonraker.
Mar 14 16:02:54 raspberrypi python3[8316]: Loading Service Config: {"KlipperConfigFolder": "/home/pi/printer                                    _data/config", "MoonrakerConfigFile": "/home/pi/printer_data/config/moonraker.conf", "KlipperLogFolder": "/h                                    ome/pi/printer_data/logs", "LocalFileStoragePath": "/home/pi/printer_data/octoeverywhere-store", "ServiceNam                                    e": "octoeverywhere", "VirtualEnvPath": "/home/pi/octoeverywhere-env", "RepoRootFolder": "/home/pi/octoevery                                    where"}
Mar 14 16:02:54 raspberrypi python3[8316]: Config file doesn't exist. Creating a new file now!
Mar 14 16:02:54 raspberrypi python3[8316]: 2023-03-14 16:02:54,974 - INFO - ################################                                    ##
Mar 14 16:02:55 raspberrypi python3[8316]: 2023-03-14 16:02:54,976 - INFO - #### OctoEverywhere Starting ###                                    ##
Mar 14 16:02:55 raspberrypi python3[8316]: 2023-03-14 16:02:54,977 - INFO - ################################                                    ##
Mar 14 16:02:55 raspberrypi python3[8316]: 2023-03-14 16:02:54,978 - INFO - Plugin Version: 2.3.13
Mar 14 16:02:55 raspberrypi python3[8316]: 2023-03-14 16:02:54,979 - INFO - Our update config include was no                                    t found in the moonraker config, so we added it.
Mar 14 16:02:55 raspberrypi python3[8316]: 2023-03-14 16:02:54,980 - INFO - No update config found or it was                                     out of date, writing a new file.
Mar 14 16:02:55 raspberrypi python3[8316]: 2023-03-14 16:02:54,980 - INFO - No config file was found on disk                                    , so we are going to attempt to restart moonraker.
Mar 14 16:02:55 raspberrypi python3[8316]: 2023-03-14 16:02:55,974 - INFO - No printer id was found, generat                                    ing one now!
Mar 14 16:02:56 raspberrypi python3[8316]: 2023-03-14 16:02:56,199 - INFO - New printer id created: 1D9OYPVK                                    H1SJ2G50DWH0HXPAIC42BVGA33BG1GPEOS9EANZ39OLAJPEPAB01
Mar 14 16:02:56 raspberrypi python3[8316]: 2023-03-14 16:02:56,654 - INFO - No private key was found, genera                                    ting one now!
Mar 14 16:03:05 raspberrypi python3[8316]: 2023-03-14 16:03:05,542 - INFO - New private key created.
Mar 14 16:03:05 raspberrypi python3[8316]: 2023-03-14 16:03:05,544 - INFO - We found our name existing in th                                    e moonraker allowed service file, so there's nothing to do.
Mar 14 16:03:05 raspberrypi python3[8316]: 2023-03-14 16:03:05,545 - INFO - MDns cache file reset
Mar 14 16:03:05 raspberrypi python3[8316]: 2023-03-14 16:03:05,780 - INFO - Setting up relay with frontend p                                    ort 80
Mar 14 16:03:05 raspberrypi python3[8316]: 2023-03-14 16:03:05,781 - INFO - OctoPingPong stats reset
Mar 14 16:03:05 raspberrypi python3[8316]: 2023-03-14 16:03:05,783 - INFO - PingPong has detected a first ti                                    me run. Updating latency stats now.
Mar 14 16:03:33 raspberrypi python3[8316]: 2023-03-14 16:03:33,507 - INFO - Attempting to talk to OctoEveryw                                    here, server con wss://starport-v1.octoeverywhere.com/octoclientws[1] wsId:1975291160

Installer failed. Failed to wait for printer id; Traceback (most recent call last):
  File "/home/pi/octoeverywhere/moonraker-install-completer.py", line 138, in Run
    self.CheckIfPrinterIsConnectedToAccountAndSetupIfNot(existingPrinterId)
  File "/home/pi/octoeverywhere/moonraker-install-completer.py", line 534, in CheckIfPrinterIsConnectedToAcc                                    ountAndSetupIfNot
    raise Exception("Failed to wait for printer id")
Exception: Failed to wait for printer id

### What did you expect to happen

For the installer to complete without errors.

### How to reproduce

Start the OctoEverywhere installer from a fresh Raspbian Image through Kiauh

th33xitus, 2023-03-14:
@QuinnDamerell Any idea on this one? Did something in the OE installer changed recently?

QuinnDamerell, 2023-03-14:
@Endracion, thanks for reporting this! 

Can you tell me what data path the moonraker instance is using? Is it `/home/pi/printer_data/logs`?

Endracion, 2023-03-14:
@QuinnDamerell It was all default from kiauh, so yes `/home/pi/printer_data/logs` I just confirmed.

QuinnDamerell, 2023-03-15:
@Endracion thanks for that info! 

Can you edit the file moonraker-install-completer.py  in ~/octoeverywhere/ and change DebugLogging the True? Then run the install from kiauh again and send me the output please!

QuinnDamerell, 2023-03-15:
@Endracion, that's really interesting. Do you know what messed up about the moonraker config? 

I shouldn't have been a server issue, that ID it's looking for is just in the local config file after the service starts. It does require the service to start to get it, so I wonder if the service was failing for some reason. But if it was, it would be in the logs, and the logs look fine.

That's really weird... I wish I could figure this out. Thanks for your help, I will keep an ear out for other users who hit anything similar.

Endracion, 2023-03-15:
@QuinnDamerell It had written some OctoEverywhere references in the moonraker.conf, but the service wasn't actually starting properly, and even after uninstalling, the OctoEverywhere references remained in moonraker.conf and were throwing warnings in the UI, so I went into both moonraker and the bkp conf and removed all mentions of OctoEverywhere, which did away with the warnings.

QuinnDamerell, 2023-03-15:
@Endracion do you happen to the broken config file around or the error messages you were seeing in the UI?

Endracion, 2023-03-15:
@QuinnDamerell Sorry, I do not, I only recall it mentioning warnings regarding OctoEverywhere and that it was also looking in the bkp file.

QuinnDamerell, 2023-03-15:
No problem, I was just wondering what the issue was so I could look into it. 

Thanks for all of the info and discussion, if you have any other issues please reach out!

closed 2023-03-15 by Endracion