dw-0/kiauh #314: a way to force kiauh not to use port 80, so I could easily install a framed wrapper

piotrb, 2023-03-15:
### Is your feature request related to a problem? Please describe

The problem really is I'd like to easily navigate between all the different tools installed on the host .. there is the klipper frontend .. but I also am running grafana/prometheus to get detailed metrics from my printer visible (see https://github.com/piotrb/klipper-stats if that's something you'd like to make easy for folks to install .. may be a bit heavy weight to run on anything other than a Pi4)

Any time I want to get to grafana I need to go to the address bar and edit the url .. add port 3000 .. etc its a bit of a drag 

### Describe the solution you'd like

I was thinking it may be nice to make the page on port 80 just a static simple html page with a fixed bit of navigation, and essentially an old school html frameset .. this way the nav can stay on top .. and the UI can get loaded below .. and can easily switch between them ... but Kiauh really doesn't want to let go of port 80 unless the other frontend is installed .. 

So I'm just hoping for there being a way to perhaps either "install" a custom frontend as one of the "main" ones .. so that then when fluidd/mainsail install they'd detect that and make you choose a port .. (actually this probably the easiest option .. a custom frontend .. you select a folder .. and it sets you up with maybe the nginx config file and a www-root folder off of the user's home .. 

This way that can be customized as needed .. and counts as something "blocking" port 80 .. 

### Describe alternatives you've considered

well I read through the code .. its quite specific around how it detects the alternate port stuff .. I guess I could just modify things myself in /etc/nginx .. but I wanted something that wouldn't get reverted any time I re-installed the frontend .. it kinda nice to be able to uninstall/reinstall fluidd and have it not miss a  beat ..

th33xitus, 2023-03-18:
Mhhhh... what about always displaying the port selection dialog even if you're currently installing the very first klipper interface? which would currently just default to port 80. but maybe rewriting the dialog might make sense, for some more customizing options upfront, just like you would require. the dialog could be easily "pre-filled" with port 80, and users that are unsure should just hit enter when asked. I must confess, im not really proud of the port detection stuff at all, its so old it could require some rework, but im currently not in the mood of touching it because...well...it works :D

piotrb, 2023-03-18:
lol .. I like it .. if that's a quick fix .. I'll take it .. it'd solve the problem .. because its super hard to force the port selection right now :)

Tam-Lin, 2023-03-29:
As a suggestion (or maybe this could be a different issue, which I was thinking about how to handle anyway), instead of choosing the port selection dialog, why not add another layer of hostname?  ie fluidd.printername.local, and mainsail.printername.local, and so on?  That saves you from having to remember the port at all.

th33xitus, 2023-03-29:
@Tam-Lin there was a discussion about this here: https://github.com/th33xitus/kiauh/issues/113
And afaik on mobile devices you're not able to connect to `<hostname>.local` adresses anyways.

dw-0, 2024-08-31:
Im sorry it took so long but this should now be possible with KIAUH v6 which's alpha got released just a moment ago.
It has it's own config file now (see the changelog), where you can specify the port of both Mainsail and Fluidd. If those are defined, no dialog pops up during installation and KIAUH will simply use the specified port.

closed 2024-08-31 by dw-0

gusarg81, 2024-09-03:
Hi, is possible to change the web port on already installed KIAUH? Since I need Apache to run on port 80 (if not, then is safe to change it manually on Nginx conf?).

dw-0, 2024-09-03:
hi, no changing ports through kiauh is not possible (would be a usefull feature though i guess).
but it is safe to change it manually in the nginx conf. stop the nginx service. change the port, then restart the nginx service again.