dw-0/kiauh #326: Mainsail install cannot find printers.

CorruptName, 2023-04-06:
### Linux Distribution

Raspberry Pi OS 32-bit (latest, fully updated, bullseye)

### What happened

After clean install I Installed 4 instances of Klipper & Moonraker.
Next, installed Fluidd and added all 4 printers.
Next, installed Mainsail & cannot add printers.
Only able to add 1 printer if I give the address "klipper.local:9000" (which is the port which mainsail is attached to)
But if I use the same addresses used for Fluidd "klipper.local:7125-7128" Mainsail cannot find printers.
I have also tried using IP directly as in: "192.168.0.96:7125" - no help.

### What did you expect to happen

I expected mainsail to find printers when given the correct address & port

### How to reproduce

described above

closed 2023-04-06 by th33xitus

CorruptName, 2023-04-06:
how is this "not kiauh"?
Did kiauh not handle the configuration of the installation?
No response, no F you, just close and pretend it has nothing to do with kiauh?
At the very least kiauh should inform the user that he cannot install both Mainsail & Fluidd when using multiple instances.
But I think it's possible just misconfigured (by kiauh)

th33xitus, 2023-04-06:
> Did kiauh not handle the configuration of the installation?

Not the kind of configuration you describe. It configures NGINX, nothing more. All other configurations like adding the printers to the interfaces is up to the user.

CorruptName, 2023-04-06:
> > Did kiauh not handle the configuration of the installation?
> 
> Not the kind of configuration you describe. It configures NGINX, nothing more. All other configurations like adding the printers to the interfaces is up to the user.

Then I think you misunderstand the issue. Perhaps I'm not being clear.
There is something within the configuration which is preventing the secondary UI (Be it Mainsail or Fluidd, whichever is installed second) from adding the same printers which are added to the first UI. Exact same addresses, obviously using the same moonraker.conf. It seems as if moonraker is not able to accept connection from both Mainsail & Fluidd at the same time. How could it even be a CORs issue if the other ui is actively working?

CorruptName, 2023-04-06:
If you doubt that this is an issue, please try it:
Install multiple instances of klipper, moonraker (I think anything more than 1 will have the issue but I used 4)
then install Mainsail and Fluidd and try to add the printers to both.

CorruptName, 2023-04-06:
I got it working by editing:

/etc/nginx/nginx.conf

from:
```
events {
        worker_connections 768;
        #multi_accept on;
}
```
to:
```
events {
        worker_connections 768;
        #multi_accept on;
}
```
and moonraker.conf (in each instance) adding:
```
[authorization]
cors_domains:
    *.local:*
```

th33xitus, 2023-04-06:
> If you doubt that this is an issue, please try it

[image]

CorruptName, 2023-04-06:
what does moonraker.conf [authorizations] look like?

th33xitus, 2023-04-06:
[image]

> did you modify it from stock (kiauh)?

of course not. hence my whole point of marking this issue as: not kiauh

CorruptName, 2023-04-06:
This is very strange. I have literally done 5 clean installs today only using kiauh and changing nothing (until it didn't work)
and even at this very moment if I remove *.local:* connections break.

th33xitus, 2023-04-06:
of course it does break. that is my whole point. you need to configure it in your moonraker.conf.
but that is NOT in the scope of kiauhs installer.

CorruptName, 2023-04-06:
but didn't you say "of course not" when I asked if you had to modify your moonraker.conf?

th33xitus, 2023-04-06:
yes, because i didn't need to modify my moonraker.conf.
pay attention how i open the interfaces. by ip, not by hostname.

CorruptName, 2023-04-06:
but I have tried the same and still won't connect. I'm also confused why the first UI connects without modification but the second one won't?

CorruptName, 2023-04-06:
It's obviously completely up to you how you view this but if your goal with Kiauh is to make things smooth, a whole day of troubleshooting was not very smooth for me. I get that most people will only use one frontend but noobs like me need to play with both until they decide. I am still very grateful for the hard work put into this awesome tool. just trying to help make it better.

th33xitus, 2023-04-06:
As i already said, it typically works if you configure the moonraker.conf correctly.
And as you can see, my try worked perfectly ootb. What should i change in KIAUH in your opinion to "make it better" if i can't even reproduce it? Using ".local" addresses can often lead to issues, hence i wouldn't recommend using them at all although they might be comfortable. Wont work on mobiles for example anyways...

CorruptName, 2023-04-06:
It works perfectly fine on my mobiles. I think local hostnames just don't work well with windows.
I can't really tell you what to change since I still don't quite understand why the first installed frontend works perfect with all instances already added, but the second frontend install fails to add and refuses to let me add even with direct ip addresses.
But I guess that's what I would suggest, if you can make the second installation add the instances like the first one does.

th33xitus, 2023-04-06:
> But I guess that's what I would suggest, if you can make the second installation add the instances like the first one does.

You have to add the instances in Fluidd/Mainsail, thats out of KIAUHs scope.
KIAUH doesn't add ANY instances to ANY interface. In both interfaces you have to configure the instances.