dw-0/kiauh #335: Kiuah without Systemd support

Mate76PL, 2023-04-26:
### Is your feature request related to a problem? Please describe

Kiuah is often installed on old terminals. I'm installing on a Dell Wyse3040. You have to search a lot to find a version of Linux that is small enough to fit on an 8Gb emmc. The best in my opinion is AntiX. It is very light. Unfortunately, it does not support Systemd, so some Kiuah scripts get stuck. Is it possible to execute scripts without SystemD support?

### Describe the solution you'd like

Scripts without systemd

th33xitus, 2023-04-26:
Please see: https://github.com/th33xitus/kiauh/issues/287

The-Monkey-King, 2023-07-27:
I mean, it's going to barf, or at least stop, when it tries to run Klipper. And that's because Klipper is a service that uses systemd. You cannot get around it, period. It's what stopped me from using earlier versions of Windows Subsystem for Linux (WSL) because it did not employ systemd.

If anyone can rewrite Klipper from scratch to not use it, I am all ears/eyes.

closed 2023-10-31 by dw-0