dw-0/kiauh #336: When using gcode_shell_command some basic commands return error

horatio1989, 2023-04-30:
### Linux Distribution

klipper
v0.11.0-190-g19877425

mainsail-config
v1.0.0-11-ge4976f6

moonraker
v0.8.0-39-g6c7dfe5

mainsail
v2.5.1

### What happened

I have Klipper and Mainsail on a raspberry pi 4 working. I connected to the pi via SSH and set up a git repository where all my cfg files are. In this way, I can always enter there an upload the files to git in order to have a backup. That is also working, I set the project with SSH too, so I don't need to enter the credentials every time.

Now I am writing macros to update on git my config files via a macro. I added the "gcode_shell_command" script via KIAUH and I actually can set some commands and having them working, for example I have set "ls", "pwd", "git commit -m "text"", and they worked! but by default the folder on which the terminal is, is not where the cfg files are. When I set a gcode_shell_command for "cd ~/printer_data/config/", or "cd .." I get the error "Error running command {name of the command}". Does anyone have any idea what is happening? what am I doing wrong?

Here is my klippy.log and the shell_commands.cfg

[klippy.log](https://github.com/th33xitus/kiauh/files/11361800/klippy.log)
[11_shell_commands.txt](https://github.com/th33xitus/kiauh/files/11361801/11_shell_commands.txt)

### What did you expect to happen

I expect to use all the commands that I will usually use on linux terminal via the "gcode_shell_command"

### How to reproduce

Create a simple shell command like:

```
[gcode_shell_command go_to_folder]
command: cd ~/printer_data/config/
timeout: 2.
verbose: True
```

and then run on mainsail terminal 

`RUN_SHELL_COMMAND CMD=go_to_folder`

th33xitus, 2023-04-30:
What do you even expect by running a cd (= change directory) as a macro? What is your point of just trying to navigate with a macro?

horatio1989, 2023-04-30:
if I want to commit and push the files inside "~/printer_data/config/" I need to be there. The issue is that by default, the terminal is in "~/pi/klipper/".

How can I change directory?

th33xitus, 2023-04-30:
I already wrote it in a few different issues i think. The best way to use shell macros is by calling dedicated shell scripts, not individual shell commands. There is even a full wiki page as an example: https://github.com/th33xitus/kiauh/wiki/How-to-autocommit-config-changes-to-github%3F

horatio1989, 2023-04-30:
thanks! I will take a look at that, I didn't find that page, even though I was up to the third google search page! Sorry if the issue was wrongly placed in kiauh, I thought it belonged here.

th33xitus, 2023-04-30:
No problem. Support questions are likely best placed in the klipper discord (you find me there as well). This issue tracker is for real issues and real bugs, or feature requests.

closed 2023-04-30 by th33xitus