dw-0/kiauh #342: Breaking change in crowsnest causes "Dependency: 'camera-streamer' not found. Exiting" error

dburr, 2023-05-24:
### Linux Distribution

Raspbian 11 Bullseye

### What happened

As of today it appears as though the Crowsnest folks are introducing a **[breaking change](https://github.com/mainsail-crew/crowsnest/issues/85)** that is [causing trouble for many people](https://github.com/mainsail-crew/crowsnest/issues/112#issuecomment-1559106007). I don't fully understand what is going on, but the gist of it seems to be that they have switched to a new camera streaming library `camera-streamer` which is unsupported by most distributions, and is also completely incompatible with Obico.

Currently people are being advised to stay on the `legacy/v3` branch.

To manually revert (in case you have run the upgrade) run these steps (as documented [here](https://github.com/mainsail-crew/crowsnest/issues/112#issuecomment-1559106007)):

```shell
$ cd crowsnest
$ git fetch && git pull
$ git switch legacy/v3
$ make install
$ sudo systemctl restart crowsnest.service
```

### What did you expect to happen

Camera works as normal

### How to reproduce

Run the crowsnest upgrade

kennethjiang, 2023-05-24:
One more piece of info that may or may not be important:

There is a line that complains about "libcamera.so.0". However, I checked and everything seems to be working fine. Webrtc/mp4, and mjpeg.

```
pi@raspberrypi:~/moonraker-obico $ systemctl status crowsnest
● crowsnest.service - crowsnest - Multi Webcam/Streamer Control Deamon
     Loaded: loaded (/etc/systemd/system/crowsnest.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-05-24 11:55:19 PDT; 10min ago
       Docs: https://github.com/mainsail-crew/crowsnest
   Main PID: 682 (crowsnest)
      Tasks: 24 (limit: 4915)
        CPU: 20.305s
     CGroup: /system.slice/crowsnest.service
             ├─ 682 /bin/bash /usr/local/bin/crowsnest -c /home/pi/printer_data/config/crowsnest.conf
             ├─ 992 /bin/bash /usr/local/bin/crowsnest -c /home/pi/printer_data/config/crowsnest.conf
             ├─1048 xargs /home/pi/crowsnest/bin/camera-streamer/camera-streamer
             ├─1049 /bin/bash /usr/local/bin/crowsnest -c /home/pi/printer_data/config/crowsnest.conf
             ├─1050 /home/pi/crowsnest/bin/camera-streamer/camera-streamer --http-port=8080 --camera-path=/dev/video0 --camera-type=v4l2 --camera-format=MJPG --camera-snapshot.height=1080 ->
             └─1752 sleep 120

May 24 11:55:23 raspberrypi crowsnest[682]: INFO: Found 1 total available Device(s)
May 24 11:55:23 raspberrypi crowsnest[914]: libcamera-hello: error while loading shared libraries: libcamera.so.0: cannot open shared object file: No such file or directory
May 24 11:55:23 raspberrypi crowsnest[682]: INFO: Found 1 available v4l2 (UVC) camera(s)
May 24 11:55:23 raspberrypi crowsnest[927]: /dev/v4l/by-id/usb-046d_0825_124E68D0-video-index0 -> /dev/video0
May 24 11:55:23 raspberrypi crowsnest[927]: Supported Formats:
May 24 11:55:23 raspberrypi crowsnest[927]: Supported Controls:
May 24 11:55:23 raspberrypi crowsnest[682]: Try to start configured Cams / Services...
May 24 11:55:25 raspberrypi crowsnest[682]: INFO: Configuration of Section [cam 1] looks good. Continue ...
May 24 11:55:27 raspberrypi crowsnest[682]:  ... Done!
May 24 11:55:27 raspberrypi crowsnest[992]: Starting camera-streamer with Device /dev/video0 ...
```

meteyou, 2023-05-24:
@dburr ustreamer is also supported with the new version. you are only able to switch to camera-streamer if you want it in your config. but the v4 version is incompatible with buster, because ustreamer & camera-streamer need libcamera in the new version. so only users with a outdated OS should not update!

bahbka, 2023-05-24:
I have the same issue on my fresh raspbian install (bullseye), after Crowsnest upgrade to 4 it stopped working with error
`crowsnest: Dependency: 'camera-streamer' not found. Exiting!`
So I'd roll back Crowsnest to 3.

meteyou, 2023-05-24:
@bahbka then you dont follow the update guide. you have to do it this way:

```
cd ~/crowsnest
make uninstall

# update Crowsnest to v4

cd ~/crowsnest
sudo make install
```

without this, nobody will have `camera-streamer` installed...

bahbka, 2023-05-25:
@meteyou KIAUH had not provide this update guide or I had missed it. It helps with `'camera-streamer' not found`, but now there is new error `ERROR: No usable Devices Found. Stopping crowsnest.` :( Trying to roll back to 3 and stick with it.

meteyou, 2023-05-25:
@bahbka that was the reason for the breaking change announcement on github since 02/2023 and the announcement on the mainsail discord.

tastyratz, 2023-06-02:
> @bahbka then you dont follow the update guide. you have to do it this way:
> 
> ```
> cd ~/crowsnest
> make uninstall
> 
> # update Crowsnest to v4
> 
> cd ~/crowsnest
> sudo make install
> ```
> 
> without this, nobody will have `camera-streamer` installed...

I also experienced this with Mainsailos using update manager to install. While it wasn't a Kiauh issue, I found this bug when googling the error.
I just wanted to say this was all it took to get my c270 back online. Appreciated!

bahbka, 2023-06-02:
@tastyratz I didn't follow the update guide because KIAUH didn't provide it
I already had tried Crowsnest 4, my rpi cam works but very bad with violet blinks on the picture, tried ustream and camera-streamer. Picture with Crowsnest 3 is really good, so I decided to stick with it, because I have no time to debug this.

therealzoomgod, 2023-06-07:
> @bahbka that was the reason for the breaking change announcement on github since 02/2023 and the announcement on the mainsail discord.

I don't want to go tit for tat.  Just going to say as a retired software engineer of 40 years blaming end users for not checking in on github or discord when the script itself can update seems like a blame game and is counter productive.  It would not be difficult for the other interfaces (Mainsial/Fluuid etc) to display a message and stop before doing anything.  The update in Mainsail is what hosed mine up, 

**_Kiuah did alert me about buster being EOL so kudos for that._**

For anyone else stuck this is what it took to get mine up and running again.
 cd ~/crowsnest/
make uninstall
git fetch && git pull
git switch legacy/v3
sudo systemctl daemon-reload
make clean
make build
sudo make install
reboot Pi

Thanks

therealzoomgod, 2023-06-13:
> @therealzoomgod and then the user click on "update all" and the Webinterface have no time to be display the warning...
> 
> Your guide is also wrong for bullseye users. Old stop sharing wrong guides or without the right hint!

You know why there isn't one solution ?  Because it depends on what was done to try and resolve it after it blew up.  That is step by step to what fixed mine, the service got screwed.  Go ahead and post a AIO fix and I bet I will find a scenario it won't work for.

I'll go back and finger point to the origional problem, installers should check the OS for compatibility.  Perhaps it wasn't tested on bullseye, I don't know,  I personally had no idea bullseye was no longer supported because in Mainsail it never warned me and offers updates for it.   I used the in GUI updates, did not have web cam plugged in for some time and when I did it no longer worked.  As I said expecting people to scroll through Discord to find warnings about an OS is just insane when the GUI's could notify all users affected.

th33xitus, 2023-06-13:
In the end it all boils down to a broken Crowsnest update causing all those headaches. Thats the only fingerpoint that should be made. It's crazy to expect regular users to check support discords or even github issues for any announcements of updates causing breaking changes. Bad decisions were made and a broken or lets say a non forward looking update was released, that's it. And its not in the obligations to show such things in Mainsail, nor in KIAUH.

closed 2023-06-13 by th33xitus