dw-0/kiauh #345: add info to Wiki

FireflyJar, 2023-05-29:
### Is your feature request related to a problem? Please describe

I couldn't figure out how to get to OctoPrint,  wasted a lot of time trying.

### Describe the solution you'd like

Please add info in wiki :    to access OctoPrint , use port :5000,   exp:   yourPiAddress.:5000

Thanks

th33xitus, 2023-05-29:
You just need to watch the terminal output after the installation:
[image]

closed 2023-05-29 by th33xitus

FireflyJar, 2023-05-29:
yup,  I need a reminder.   Did this install weeks ago and don't remember it..    few weeks I will also forget I posted this. ;)