dw-0/kiauh #356: Libjpeg62-turbo error when trying to install mobileaker

simpat1zq, 2023-06-18:
### Linux Distribution

Ubuntu x64

### What happened

packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libjpeg62-turbo is not available, but is referred to b
y another package.
This may mean that the package is missing, has been obsoleted,
 or
is only available from another source 
This may mean that the package is missing, has been obsoleted,
 or
is only available from another source

E: Package 'libjpeg62-turbo' has no installation candidate

#====================================

### What did you expect to happen

Install mobileraker

### How to reproduce

Install mobileraker. 

### Additional information

I think this may be related to Ubuntu. I just installed on my pi with no problem.

simpat1zq, 2023-06-19:
Just did a tiny bit of research, and I could very well be wrong about this, but I think that ubuntu might use libjpegturbo8 and pi distros use 62, but that they are mostly the same thing. 

But I suspect you all know a lot more about this stuff than me.

th33xitus, 2023-06-19:
> So I'm not sure if it's a kiauh issue or mobileraker.

KIAUH executes the install script mobileraker is providing. So you were pretty much correct that this issue might be related to the one you linked and the dependencies need to be covered by the Mobileraker maintainer. I assume he will work on it and after that fix is released, the implementation in KIAUH will also work.

closed 2023-08-23 by dw-0