dw-0/kiauh #357: No interface shown, "No such file or directory" error on attempt to update or quit of KIAUH

terransiladi, 2023-06-19:
### Linux Distribution

Linux mainsail 6.1.21-v7

### What happened

When I start KIAUH no interface is displayed, just a "Perform action prompt". Sometimes it asks if I want to update, if I say yes, I get the following error (still with no interface): "/=======================================================\\: No such file or directory". If I say no still no interface displayed, and when I quit, I get the exact same error.

Image here shows the error at the top, and then the commands I ran previously to get the error (error clears the screen when it occurs)

[image]

### What did you expect to happen

Interface to be displayed as normal, no error on update attempt or quitting of KIAUH.

### How to reproduce

Unfortunately I don't know exactly how to reproduce what caused my install to do this. I installed mainsail OS directly onto my pi. Only change was installing octoeverywhere, then installing KIAUH in order to uninstall Octoeverywhere (which appeared to be successful when I did it).

### Additional information

I tried deleting the KIAUH folder from my system and recloning from GIT, exact same issues occur. I haven't been able to find a KIAUH log anywhere to look at - does KIAUH create a log anywhere?

th33xitus, 2023-06-19:
Im unable to reproduce the issue.
Also i don't know what OS you are actually using. You mentioned mainsailOS but you wrote `Linux mainsail 6.1.21-v7`, which i have no idea what that should be.

terransiladi, 2023-06-19:
Sorry about that, I copied that out of terminal without thinking about it, that's the kernal info. OS info is Mainsail OS 1.2.1, which is built on Raspbian GNU/Linux 11 (bullseye).

closed 2023-08-23 by dw-0