dw-0/kiauh #358: Mobilraker's Companion Install Fails on Debian Bookworm

DGEllis, 2023-06-23:
### Linux Distribution

Fresh installation of Debian Bookworm on PC Engines APU4

### What happened

/=======================================================\
|     ~~~~~~~~~~~ [ Installation Menu ] ~~~~~~~~~~~     |
|-------------------------------------------------------|
|  You need this menu usually only for installing       |
|  all necessary dependencies for the various           |
|  functions on a completely fresh system.              |
|-------------------------------------------------------|
| Firmware & API:          | 3rd Party Webinterface:    |
|  1) [Klipper]            |  6) [OctoPrint]            |
|  2) [Moonraker]          |                            |
|                          | Other:                     |
| Klipper Webinterface:    |  7) [PrettyGCode]          |
|  3) [Mainsail]           |  8) [Telegram Bot]         |
|  4) [Fluidd]             |  9) [Obico for Klipper]    |
|                          | 10) [OctoEverywhere]       |
|                          | 11) [Mobileraker]          |
| Touchscreen GUI:         |                            |
|  5) [KlipperScreen]      | Webcam Streamer:           |
|                          | 12) [Crowsnest]            |
|-------------------------------------------------------|
|                       B) « Back                       |
\=======================================================/
####### Perform action: 11
/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/
[✓ OK] No config directory found! Skipping backup ...

###### Checking for the following dependencies:
● wget
● curl
● unzip
● dfu-util
[✓ OK] Dependencies already met!

###### Cloning Mobileraker's companion from https://github.com/Clon1998/mobileraker_companion.git ...
Cloning into '/home/dge/mobileraker_companion'...
remote: Enumerating objects: 465, done.
remote: Counting objects: 100% (91/91), done.
remote: Compressing objects: 100% (44/44), done.
remote: Total 465 (delta 61), reused 66 (delta 47), pack-reused 374
Receiving objects: 100% (465/465), 387.83 KiB | 2.44 MiB/s, done.
Resolving deltas: 100% (269/269), done.

###### Installing Mobileraker's companion ...
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm-updates InRelease
Hit:3 http://security.debian.org/debian-security bookworm-security InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libtiff5

#=======================================================#
 Mobileraker's companion installation failed!
#=======================================================#

dge@debian:~$

### What did you expect to happen

I expected Mobileraker's Companion to install.

### How to reproduce

Try to install Mobileraker's Companion on Debian Bookwork on PCEngines APU4

### Additional information

Klipper, Moonraker, Mainsail and Crowsnest install successfully.

DGEllis, 2023-06-23:
Apparently Bookworm has libtiff6 instead of libtiff5.

DGEllis, 2023-06-23:
See https://packages.debian.org/bookworm/libtiff-tools

th33xitus, 2023-06-23:
Pretty much the same as in #356 applies here.

closed 2023-08-23 by dw-0