dw-0/kiauh #363: Warnings after installing Mobileraker

XoBoR, 2023-07-13:
### Linux Distribution

debian

### What happened

After installing Mobileraker, I get the warning: 

> [update_manager mobileraker]: Moonraker is not permitted to restart service 'mobileraker'. To enable management of this service add mobileraker to the bottom of the file /home/rob/printer_7_data/moonraker.asvc. To disable management for this service set 'is_system_service: False' in the configuration for this section.

Removing Mobileraker does not remove the update manager entry currently, so the warning needs to be fixed manually. 

### What did you expect to happen

I did not expect a warning after the installation process and I did expect the update manager entries to be removed on uninstall

### How to reproduce

install mobileraker..

AndreKR, 2023-09-24:
I'm having the same problem.

I installed "Mobileraker" because I thought I need it to use Mobileraker. (I learned later that it actually install Mobileraker *companion* which is only needed for notifications.)

I removed it again and I got the same message as you got. Then I rebooted and now I'm getting this:

[image]

ElvinNoel, 2024-01-09:
[image]

I'm having the same issues. Just recently set up a mutli instance klipper. Installed mobileraker and encounter this issue.

shodanx2, 2024-04-23:
Hi,

Ran maintenance on my klipper controller today

I noticed the following error messages related to mobileraker

```
Moonraker warning

[update_manager]: Failed to load extension mobileraker: Invalid path for option `install_script` in section [update_manager mobileraker]: Path `/home/user/mobileraker_companion/scripts/install-mobileraker-companion.sh` does not exist
Moonraker warning

Unparsed config option 'origin: https://github.com/Clon1998/mobileraker_companion.git' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'primary_branch: main' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'managed_services: mobileraker' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
```

Since I have not ended up using mobileraker, I will be removing it.  Just wanted to let you know of this issue. 

That is an issue before uninstallation BTW. 

I am removing it as it is a cloud dependency as far as I can tell it require the app maintainers' servers to work. 

After uninstallation and reboot got the following notifications

```
Moonraker warning

Server configuration error: No files matching include directive [include octoeverywhere-system.cfg]
Loaded server from most recent working configuration: '/home/user/printer_1_data/config/.moonraker.conf.bkp'
Please fix the issue in moonraker.conf and restart the server.
Moonraker warning

[update_manager]: Failed to load extension mobileraker: Invalid path for option `path` in section [update_manager mobileraker]: Path `/home/user/mobileraker_companion` does not exist
Moonraker warning

[update_manager]: Failed to load extension octoeverywhere-1: Invalid path for option `path` in section [update_manager octoeverywhere-1]: Path `/home/user/octoeverywhere` does not exist
```

After another reboot, getting these notifications now
```

Moonraker warning

Server configuration error: No files matching include directive [include octoeverywhere-system.cfg]
Loaded server from most recent working configuration: '/home/user/printer_1_data/config/.moonraker.conf.bkp'
Please fix the issue in moonraker.conf and restart the server.
Moonraker warning

[update_manager]: Failed to load extension mobileraker: Invalid path for option `path` in section [update_manager mobileraker]: Path `/home/user/mobileraker_companion` does not exist
Moonraker warning

[update_manager]: Failed to load extension octoeverywhere-1: Invalid path for option `path` in section [update_manager octoeverywhere-1]: Path `/home/user/octoeverywhere` does not exist
Moonraker warning

Unparsed config option 'origin: https://github.com/Clon1998/mobileraker_companion.git' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'primary_branch: main' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'managed_services: mobileraker' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'env: /home/user/mobileraker-env/bin/python' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'requirements: scripts/mobileraker-requirements.txt' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'install_script: scripts/install-mobileraker-companion.sh' detected in section [update_manager mobileraker]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'origin: https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere.git' detected in section [update_manager octoeverywhere-1]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'env: /home/user/octoeverywhere-env/bin/python' detected in section [update_manager octoeverywhere-1]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

[... cut ...]
Unparsed config option 'install_script: install.sh' detected in section [update_manager octoeverywhere-1]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Moonraker warning

Unparsed config option 'managed_services:
octoeverywhere-1' detected in section [update_manager octoeverywhere-1]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
```

I will report back if I get a fix. Might just wipe and reinstall fresh however !

dw-0, 2024-08-31:
This should be fixed with KIAUH v6 which's alpha got released just a minute ago.

closed 2024-08-31 by dw-0