dw-0/kiauh #365: update_system: command not found using update a) [update all]

jgefaell, 2023-07-22:
### Linux Distribution

biqu@BTT-CB1:~$ uname -a
Linux BTT-CB1 5.16.17-sun50iw9 #2.3.2 SMP Fri Mar 3 17:05:59 HKT 2023 aarch64 GNU/Linux
biqu@BTT-CB1:~$ cat /etc/*release*
# PLEASE DO NOT EDIT THIS FILE
BOARD=h616
BOARD_NAME="BTT-CB1"
DISTRIBUTION_CODENAME=bullseye
VERSION=2.3.2
LINUXFAMILY=sun50iw9
ARCH=arm64
INITRD_ARCH=arm64
BRANCH=current
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

### What happened

Selecting a) [update all] when 12) [System] has updates results in update_system: command not found using update a) [update all]

If other packages require updates they will be installed.

### What did you expect to happen

Selecting a) [Upate all] is selected the [System] updates should be applied.

### How to reproduce

Select a) [Update all] when 12) [System] has updates.

KIAUH installed on top of BTT-CB1 distro.

### Additional information

Specifically selecting 12) [System] is successful.

th33xitus, 2023-07-22:
I just pushed a fix, i assume you already have your system up to date now, so you can't try and test if its working for you now?
I tested it on my dev system and it seemed to work again.

jgefaell, 2023-07-22:
I have updated all components. Next time there is a system update I'll report back. Is that good?

Thank you for your prompt attention. :)

th33xitus, 2023-07-22:
That's totally fine :)

NovaViper, 2023-07-26:
Hey it works! Just tested the system update script and it ran successfully

th33xitus, 2023-07-26:
thank you for the feedback! :) in that case i will close this issue.

closed 2023-07-26 by th33xitus