dw-0/kiauh #366: Cannot get camera to work on Octoprint

joelstitch, 2023-07-25:
### Linux Distribution

Raspbian  Bullseye

### What happened

Cannot get any of the printers to show the PiCam connected to the Raspberry Pi

### What did you expect to happen

I expected the video feed to show up on at least one of the octoprint instances

### How to reproduce

Try to see the camera contents on Octoprint

th33xitus, 2023-07-26:
Please elaborate what the issue is related to KIAUH.

joelstitch, 2023-07-26:
Not sure what the issue is related to honestly. When I run Octoprint by
itself the PiCam works fine but If I use a Octoprint installed through
Kiauh the Picam does not work.

th33xitus, 2023-07-26:
> When I run Octoprint by itself the PiCam works fine

Octoprint pre installed on Octopi? In that case a required camera streamer is probably installed as well. That doesn't happen during the Octoprint installation with KIAUH. It's an assumption of what you might want to say, though im absolutely not sure if thats what you actually try to say.

joelstitch, 2023-07-26:
Must be the streamer then. I'll do some research on how to add it and update the issue on github.

wolfnacht44, 2023-08-22:
@joelstitch 
when installing octoprint with kiauh, ive had to install mjpeg-streamer, and point webcam to http://octoprintip:8080/?action=stream to use webcam

I vaugely remember having to raspi-config and enabling legacy support as well using raspi official HQ camera
double check your driver setting under /boot/config.txt as well if using libcamera/camera-streamer
[driver info found here
](https://www.raspberrypi.com/documentation/computers/camera_software.html)

there may be more configuration in mjpeg-streamer

sidenote: 
Also Octopi ships with mjpeg-streamer installed and configured
if youre installing octopi from the raspi imager then I believe you're using a different fork with camera-streamer stack
Refer to [THIS](https://octoprint.org/blog/2023/05/24/a-new-camera-stack-for-octopi/) blog post

closed 2023-10-31 by dw-0