dw-0/kiauh #369: Webui install failure, nginx error

xMxMiller, 2023-08-07:
### Linux Distribution

Armbian 21.11.0-trunk Bullseye

### What happened

Tryed to install Mainsail and install failed when attempting to start service.
Tryed Fluid, same issue.
[webui install failure.txt](https://github.com/th33xitus/kiauh/files/12283279/webui.install.failure.txt)

### What did you expect to happen

To be brought back to installation menu page after the webui had been installed.

### How to reproduce

Follow install guide.

### Additional information

Mellow gemini Fly v3 control board I got in a Siboor v0.2 r1 kit.

th33xitus, 2023-08-07:
The issue is the following:
```
Aug 07 16:57:12 voron-0 nginx[2084]: nginx: [emerg] open() "/etc/nginx/sites-enabled/klipper_webui" failed (2: No such file or directory) in /etc/nginx/nginx.conf:60
```
That is no file that gets installed with KIAUH.
You say that you're using a Fly Gemini, so are you using their Image as well? Afaik they have both interfaces already installed and you select one depending on which you have activated in a special config file. I don't know which it was from the top of my head, but they do it kinda differently than KIAUH would install and configure NGINX for the use of Mainsail or Fluidd.

xMxMiller, 2023-08-07:
Cheers for looking at this,

I am running the image it had pre-installed and wanted to use kiauh to make sure everything was upto date aswell as change from Fluid to Mainsail.
Im about to re-image to the original FLYOS.

th33xitus, 2023-08-07:
Check out the following guide:
https://3dpandme.com/2022/07/11/mellow-fly-gemini-v2-installation-guide-for-voron-0-1/

And search for "Swapping to Mainsail" in this guide. This explains what file you need to edit, to switch from Fluidd to Mainsail and vice versa. Using KIAUH on their image to try and uninstall the one, and installing the other won't work. They somehow manage to overwrite everything during the next boot and revert all changes made iirc.

xMxMiller, 2023-08-07:
Thanks for all your help even though my issue was unrelated to your project

closed 2023-08-07 by th33xitus