dw-0/kiauh #370: Failed to install klipperscreen

san3sann, 2023-08-09:
### Linux Distribution

 v5.0.0-52 

### What happened

An error occurred during the installation of klipperscreen, indicating that the website file does not exist,

### What did you expect to happen

You can provide a source or mirror URL where these files can be installed to ensure normal installation

### How to reproduce

These are the source links I am currently using

deb https://mirrors.ustc.edu.cn/debian/ bullseye main non-free contrib
deb-src https://mirrors.ustc.edu.cn/debian/ bullseye main non-free contrib
deb https://mirrors.tuna.tsinghua.edu.cn/debian/ bullseye main contrib non-free
deb-src https://mirrors.tuna.tsinghua.edu.cn/debian/ bullseye main contrib non-free
deb https://mirrors.tuna.tsinghua.edu.cn/debian/ bullseye-updates main contrib non-free
deb-src https://mirrors.tuna.tsinghua.edu.cn/debian/ bullseye-updates main contrib non-free
deb https://mirrors.tuna.tsinghua.edu.cn/debian/ bullseye-backports main contrib non-free
deb-src https://mirrors.tuna.tsinghua.edu.cn/debian/ bullseye-backports main contrib non-free
deb https://mirrors.tuna.tsinghua.edu.cn/debian-security bullseye-security main contrib non-free
deb-src https://mirrors.tuna.tsinghua.edu.cn/debian-security bullseye-security main contrib non-free
deb http://mirrors.aliyun.com/debian/ bullseye main non-free contrib
deb-src http://mirrors.aliyun.com/debian/ bullseye main non-free contrib
deb http://mirrors.aliyun.com/debian-security/ bullseye-security main
deb-src http://mirrors.aliyun.com/debian-security/ bullseye-security main
deb http://mirrors.aliyun.com/debian/ bullseye-updates main non-free contrib
deb-src http://mirrors.aliyun.com/debian/ bullseye-updates main non-free contrib
deb http://mirrors.aliyun.com/debian/ bullseye-backports main non-free contrib
deb-src http://mirrors.aliyun.com/debian/ bullseye-backports main non-free contrib
deb http://mirrors.163.com/debian bullseye main non-free

### Additional information

错误:1 http://repo.mobian-project.org unstable/main arm64 libglib2.0-dev-bin arm64 2.68.1-2mobian1
  404  Not Found [IP: 91.121.79.211 80]
错误:2 http://repo.mobian-project.org unstable/main arm64 libglib2.0-bin arm64 2.68.1-2mobian1
  404  Not Found [IP: 91.121.79.211 80]
错误:3 http://repo.mobian-project.org unstable/main arm64 libglib2.0-0 arm64 2.68.1-2mobian1
  404  Not Found [IP: 91.121.79.211 80]
错误:4 http://repo.mobian-project.org unstable/main arm64 libpixman-1-dev arm64 0.40.0-1mobian1

pitamah, 2023-08-11:
Try changing bullseye to bookworm (this is the new repo)

san3sann, 2023-08-12:
Thanks, I think I know how to fix it

closed 2023-08-23 by dw-0