dw-0/kiauh #372: Add option to install in other place than $HOME.

GAZ082, 2023-08-18:
### Is your feature request related to a problem? Please describe

First time suing KIAUH and I was shocked to find out that all the stuff is installed in the $HOME. This may be ok if the user is using the Pi or SBC just for Klipper, but if not, it's not very elegant.

### Describe the solution you'd like

Add an option that when selected allows to set the path of the destination.

### Describe alternatives you've considered

You could also put the destination when executing the script: `./kiauh.sh /path/to/destination`

RichardBronosky, 2023-11-22:
@dw-0 is work on this tracked somewhere? I'd like to contribute. I don't want to put the work into creating a beautiful Pull Request if the effort is redundant and likely inferior to that of more knowledgeable project members.

My thoughts on possible improvements to the current install process.
1.  If you need to create more than 1 entry into a "visible" directory (like `$HOME`), create a single directory to contain them all
2. Never create more than 1 entry (file or directory) into a users `$HOME` (but preferably 0)
3. Only create entries into `/opt` or `/usr/local` or within the "XDG Base Directory Specification"
4. Make all download/clone paths and install roots user configurable with elegant defaults

These are listed in a "ranked" order based on what I estimate to be Value/Effort. It's better to only get #1 than nothing.

Thank you for listening.

dw-0, 2023-11-22:
@RichardBronosky the only "tracking" is the assignment to the "next" milestone. Which means im not looking into it until the current version i am working on (that means v6) is released. That will still take a while. But as KIAUH will be written in Python with v6, development should be easier and that awefull bash scripting is history. I will introduce a config file, where users can configure some specific behavior of KIAUH. Probably that will be the best place to simply introduce a configurable parameter where users can overwrite the base dir for all installations KIAUH does.