dw-0/kiauh #373: Add support for installing & updating Spoolman

t3chguy, 2023-08-19:
### Is your feature request related to a problem? Please describe

Nope

### Describe the solution you'd like

Ability to install & maintain https://github.com/Donkie/Spoolman

### Describe alternatives you've considered

Doing it manually, bah

### Additional information

Has support in moonraker & Fluidd https://github.com/fluidd-core/fluidd/pull/1119

wolfnacht44, 2023-08-22:
+1 100%
also incorperate the option to use SQLite, or specify the parameters for external db I.e. mysql, postgres, cockroach

Vanilla009, 2023-11-22:
Now integrated in mainsail too

Donkie, 2023-12-29:
In the latest Spoolman update I have improved the build system a bit, so the zipped release now has everything built and ready. It now works with Moonraker's update_manager, so I imagine adding this to Kiauh should be simpler.

roykrikke, 2024-01-03:
+1 would love to see this integrated!

szafran81, 2024-05-26:
+1 here too.
Thank you.

closed 2024-08-03 by dw-0