dw-0/kiauh #375: Section 'gcode_shell_command clear_plr' is not a valid config section error

alxkos, 2023-08-22:
### Linux Distribution

VERSION=22.05.0-trunk

### What happened

Adding [gcode_shell_command] after update of klipper/mooreker error started to appear, and it stops the loading. 

### What did you expect to happen

Same as abouve

### How to reproduce

klipper  ver v0.11.0-271-g5f990f93
moonraker  ver v0.8.0-138-gfe120952

### Additional information

After updating klipper cannot use gcode_shell_command

Section 'gcode_shell_command command' is not a valid config section

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

log:

Config error
Traceback (most recent call last):
  File "/home/mks/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/mks/klipper/klippy/klippy.py", line 145, in _read_config
    pconfig.check_unused_options(config)
  File "/home/mks/klipper/klippy/configfile.py", line 300, in check_unused_options
    % (section,))
configparser.Error: Section 'gcode_shell_command clear_plr' is not a valid config section

dw-0, 2023-08-23:
Did you check if updating Klipper causes the gcode shell extension to get removed? Otherwise that error message should not appear.

closed 2023-08-24 by dw-0

Wuwi09, 2024-05-11:
> gcode shell extension

How do i know if this is the case and how do i fix it if it is?

dw-0, 2024-05-11:
@Wuwi09 if you see the error message, it indicates that the extension is not present, but corresponding config sections are.
Fixing can be done in 2 ways:
1. re-install the extension or
2. remove the config entries

Wuwi09, 2024-05-11:
> @Wuwi09 if you see the error message, it indicates that the extension is not present, but corresponding config sections are.
> 
> Fixing can be done in 2 ways:
> 
> 1. re-install the extension or
> 
> 2. remove the config entries 

Thx for the fast response. I'm sorry but I'm new to Klipper which extensions/how to I reinstall them? Sorry to bother you again.

dw-0, 2024-05-11:
start kiauh, go into the advanced menu and select 8 to install the gcode shell command extension, then restart klipper.
either via any webinterface you might have running or go back into the kiauh main menu and instead of typing any nummer, type `restart klipper` and hit enter