dw-0/kiauh #377: Moonraker installation

Paciok17, 2023-08-24:
### Linux Distribution

LinuxMINT 19 XFCE 32bit

### What happened

When I try to install moonraker with kiauh I see this information
I tried on xubuntu 18 i386 and i have same results. What i need to do to make it working right? 

### What did you expect to happen

###### Initializing Moonraker installation ...#=======================================================#
 Versioncheck failed! Python 3.7 or newer required!
 Please upgrade Python.

### How to reproduce

 When klipper was installing i saw that "###### Installing Python 3.6.9 virtual environment...."

### Additional information

I tried on xubuntu 18 i386 and i have same results. What i need to do to make it working right? 
[image]
[image]

dw-0, 2023-08-24:
By selecting python 3 in the klipper installer, the setup picked the latest python 3 version that is likely available in your distros package list. any other or newer versions of python need to be installed explicitly. and moonraker requires python 3.7, which seems to be not the latest python 3 package in your distros package list.

Paciok17, 2023-08-24:
I have updated it to version 3.8.0 and still not working 
[image]

Paciok17, 2023-08-24:
Its working but "python3-libgpiod not found"
[image]

dw-0, 2023-08-24:
That's another required dependency of Moonraker, your current distro seems to not provide a package for.

Paciok17, 2023-08-24:
Could you tell me witch of distro will be the best to install klipper, moonraker and mainsail?
I have Lenovo T60 2008.

dw-0, 2023-08-24:
Debian Bullseye or Ubuntu 20.04, both without a desktop environment preferrably.

Paciok17, 2023-08-24:
I will check tommorrow, thanks for your help

closed 2023-08-24 by Paciok17