dw-0/kiauh #381: Unmet dependencies for packages, unable to install Klipper using KIAUH

yeash810, 2023-08-31:
### Linux Distribution

Debian

### What happened

Facing a problem installing Klipper using KIAUH for my Voron 3D Printer. This is the error i receive, any ideas as to how I can go about resolving these. Very new to this.

###### Installing packages...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-doc' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-0' for regex 'libusb-1.0'
build-essential is already the newest version (12.9).
libusb-1.0-0 is already the newest version (2:1.0.24-3).
libusb-1.0-0 set to manually installed.
pkg-config is already the newest version (0.29.2-1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libncurses-dev : Depends: libtinfo6 (= 6.1+20181013-2+deb10u3) but 6.2+20201114-2+deb11u1 is to be installed
                  Depends: libncurses6 (= 6.1+20181013-2+deb10u3) but 6.2+20201114-2+deb11u1 is to be installed
                  Depends: libncursesw6 (= 6.1+20181013-2+deb10u3) but 6.2+20201114-2+deb11u1 is to be installed
 libusb-1.0-0-dev : Depends: libusb-1.0-0 (= 2:1.0.22-2) but 2:1.0.24-3 is to be installed
 python3-dev : Depends: python3 (= 3.7.3-1) but 3.9.2-3 is to be installed
               Depends: libpython3-dev (= 3.7.3-1) but it is not going to be installed
               Depends: python3.7-dev (>= 3.7.3-1~) but it is not going to be installed
 python3-distutils : Depends: python3 (< 3.8) but 3.9.2-3 is to be installed
                     Depends: python3-lib2to3 (>= 3.6.4) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
>>>>>> Installing Klipper packages failed!

### What did you expect to happen

Klipper to install successfully

### How to reproduce

Load Raspberry Pi OS Lite (32-Bit) on to the Rasp Pi 3B+

sudo apt-get update && sudo apt-get install git -y

cd ~ && git clone https://github.com/dw-0/kiauh.git

./kiauh/kiauh.sh

Install Klipper

Install Python 3.x 

Number of Klipper Instances = 1

/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/

###### Initializing Klipper installation ...

/=======================================================\
| Please select your preferred Python version.          |
| The recommended version is Python 3.x.                |
|-------------------------------------------------------|
|  1) [Python 3.x]  (recommended)                       |
|  2) [Python 2.7]  (legacy)                            |
|-------------------------------------------------------|
|                       B) « Back                       |
\=======================================================/
###### Select Python version: 11
>>>>>> Invalid Input!

###### Select Python version: 1
   [➔] Python 3.x

/=======================================================\
| Please select the number of Klipper instances to set  |
| up. The number of Klipper instances will determine    |
| the amount of printers you can run from this host.    |
|                                                       |
| WARNING:                                              |
| Setting up too many instances may crash your system.  |
|-------------------------------------------------------|
|                       B) « Back                       |
\=======================================================/
###### Number of Klipper instances to set up: 1
   [➔] Instance count: 1

###### Installing single Klipper instance ...

###### Checking for the following dependencies:
● git
[✓ OK] Dependencies already met!

###### Cloning Klipper from https://github.com/Klipper3d/klipper ...
Cloning into '/home/voron/klipper'...
remote: Enumerating objects: 51146, done.
remote: Total 51146 (delta 0), reused 0 (delta 0), pack-reused 51146
Receiving objects: 100% (51146/51146), 139.47 MiB | 4.23 MiB/s, done.
Resolving deltas: 100% (40321/40321), done.
Checking out files: 100% (2111/2111), done.
Already on 'master'
Your branch is up to date with 'origin/master'.

###### Reading dependencies...
virtualenv
python3-dev
libffi-dev
build-essential
libncurses-dev
libusb-dev
avrdude
gcc-avr
binutils-avr
avr-libc
stm32flash
libnewlib-arm-none-eabi
gcc-arm-none-eabi
binutils-arm-none-eabi
libusb-1.0
pkg-config
dfu-util

###### Installing packages...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-doc' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-0' for regex 'libusb-1.0'
build-essential is already the newest version (12.9).
libusb-1.0-0 is already the newest version (2:1.0.24-3).
libusb-1.0-0 set to manually installed.
pkg-config is already the newest version (0.29.2-1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libncurses-dev : Depends: libtinfo6 (= 6.1+20181013-2+deb10u3) but 6.2+20201114-2+deb11u1 is to be installed
                  Depends: libncurses6 (= 6.1+20181013-2+deb10u3) but 6.2+20201114-2+deb11u1 is to be installed
                  Depends: libncursesw6 (= 6.1+20181013-2+deb10u3) but 6.2+20201114-2+deb11u1 is to be installed
 libusb-1.0-0-dev : Depends: libusb-1.0-0 (= 2:1.0.22-2) but 2:1.0.24-3 is to be installed
 python3-dev : Depends: python3 (= 3.7.3-1) but 3.9.2-3 is to be installed
               Depends: libpython3-dev (= 3.7.3-1) but it is not going to be installed
               Depends: python3.7-dev (>= 3.7.3-1~) but it is not going to be installed
 python3-distutils : Depends: python3 (< 3.8) but 3.9.2-3 is to be installed
                     Depends: python3-lib2to3 (>= 3.6.4) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
>>>>>> Installing Klipper packages failed!

JonasV328, 2023-11-14:
Could you solve it?
To me it looks not like a bug in KIAUH but it seems that you have a mixture of Debian Buster and Bullseye packages installed. The old Buster packages depend on other old Buster packages but it tries to install newer Bullseye packages. Perhaps you should try a clean upgrade to Bullseye?

pyhr0, 2023-12-13:
> Could you solve it? To me it looks not like a bug in KIAUH but it seems that you have a mixture of Debian Buster and Bullseye packages installed. The old Buster packages depend on other old Buster [...]

This fixed it for me.  I added the bullseye sources and it installed without issue.

yeash810, 2023-12-13:
Hi guys

Yes solved it. Had mixed up packages installed, also the mirror it uses to
download the packages had to be changed as it was pulling the files from a
mirror whoch wasn’t online.

Thanks for the advice & for your time :)
The Voron is up & running since 3 months now.

Best,
Yash

JLVFixer, 2024-04-25:
I apologize for bothering you, but could you please explain how you resolved this? I'm experiencing the same issue and I'm struggling to figure out how to add the Bullseye repositories. If possible, could you provide a step-by-step guide or something similar? I'm fairly new to this and could use some guidance.

Thank you very much for your help!

JonasV328, 2024-04-25:
Hello Jorge,

I can try to help you, but I did not have this problem myself. I think 
it is not really related to KIAUH but rather how to clean up the 
Raspberry Pi OS packaging system after an "un-clean" upgrade.

If your intention is to install only Klipper on Raspberry Pi OS, the 
easiest way would be to flash a new "Raspberry Pi OS Lite Bookworm" 
image to your MicroSD card. Is this possible for you to do?

Kind regards,
Jonas

yeash810, 2024-04-26:
Hey Jorge,

It shows this error because it has mixed up packages installed, also the
mirror it uses to download the packages had to be changed as it was pulling
the files from a mirror which wasn’t online. These mirrors should be
updated or removed from the kiauh program because they are offline.

https://www.raspbian.org/RaspbianMirrors

Try & change the source from where it's pulling the Raspbian Repositories
to any of the sources listed in the link shared above.
Then retry using kiauh, it should work.

All the best!

Yash