dw-0/kiauh #382: Backup filename not working

Sineos, 2023-09-04:
### Linux Distribution

Armbian 23.8 Bookworm / bash shell
KIAUH: v5.0.0-59 
```
klipper@bananapim5:~/kiauh$ git log
commit d99cda544a1c5ec5a4da87313cf53065726f9533 (HEAD -> master, origin/master, origin/HEAD)
```

### What happened

Doing a Moonraker DB or config backup:

```shell
klipper@bananapim5:~/kiauh$ ll ../kiauh-backups/moonraker_databases/
total 7
drwxr-xr-x 3 klipper klipper 3488 Sep  3 22:55 230903-2255
drwxr-xr-x 3 klipper klipper 3488 Sep  4 20:14 230904-2014
klipper@bananapim5:~/kiauh$ ll ../kiauh-backups/moonraker_databases/230904-2014/
total 4
drwxr-xr-x 3 klipper klipper 3488 Sep  4 20:14 ''$'\n'
klipper@bananapim5:~/kiauh$
```

```
klipper@bananapim5:~/kiauh$ ll ../kiauh-backups/configs/230904-2022/
total 4
drwxr-xr-x 3 klipper klipper 3488 Sep  4 20:22 ''$'\n'
klipper@bananapim5:~/kiauh$
```

Seems the file's naming is somehow broken

### What did you expect to happen

Correct file name

### How to reproduce

Do either a Moonraker DB or config backup.

dw-0, 2023-09-04:
Thanks for reporting. I think i fixed it with #[0b6613e](https://github.com/dw-0/kiauh/commit/0b6613e4647accfa7d85592f0c7f3456ce397d4f). Feel free to report back if its working now as expected.

dw-0, 2023-09-04:
Ok i forgot to fix the config backup function.... hehe. So i will need to push another commit. Hold on.

dw-0, 2023-09-04:
Ok all done, backup of moonraker databases and config folders should work again.

Sineos, 2023-09-04:
Works. Thanks a lot for the quick fix. Appreciated!

closed 2023-09-04 by Sineos