dw-0/kiauh #383: Install klipper.bin to the right directory when compiling for Linux process

ploucandco, 2023-09-16:
### Is your feature request related to a problem? Please describe

I am using Kiauh v5.0.0-61 and there is no way to get klipper.bin installed from Kiauh when compiling for Linux process.
This is in order to enable RPI MCU for an accelerometer. The following needs to be implemented according to klipper documentation:

Building the micro-controller code[¶](https://www.klipper3d.org/RPi_microcontroller.html#building-the-micro-controller-code)
To compile the Klipper micro-controller code, start by configuring it for the "Linux process":

cd ~/klipper/
make menuconfig
In the menu, set "Microcontroller Architecture" to "Linux process," then save and exit.

To build and install the new micro-controller code, run:

sudo service klipper stop
make flash
sudo service klipper start

If klippy.log reports a "Permission denied" error when attempting to connect to /tmp/klipper_host_mcu then you need to add your user to the tty group. The following command will add the "pi" user to the tty group:
sudo usermod -a -G tty pi

### Describe the solution you'd like

An extra option in order to copy klipper.bin to the right directory and execute sudo usermod -a -G tty pi if needed.

### Describe alternatives you've considered

Doing it outside of Kiauh...

Sineos, 2023-10-30:
With reference to https://klipper.discourse.group/t/problem-with-simultaneous-connection-of-two-printers/11259

Installing the Host-MCU from within KIAUH would really be a valuable addition in my opinion. 
Especially for users who are running multiple instances this would be a great help, since configuring N instances indeed requires some more in depth understanding of the entire system.

Sidenote WRT multiple instances:
I must admit that I did not check in depth, but it seems that if one instance is installed there is no easy way of adding further instances.
At least, a brave click on "Install" is blocked with:

```
#=======================================================#
 At least one Klipper service is already installed:
 ➔ /etc/systemd/system/klipper.service
#=======================================================#
```

dw-0, 2023-10-31:
i think this is more or less kind of a duplicate of #231 ? we will track it in there then.

closed 2023-12-03 by dw-0