dw-0/kiauh #386: crowsnest fails to uninstall

jakebullet70, 2023-09-29:
### Linux Distribution

DietPi V8.22

### What happened

Trying to uninstall crowsnest and fails

[image]

### What did you expect to happen

The software to uninstall

### How to reproduce

I just went through the menus to remove software, Fluid and mainsail uninstalled perfectly. crowsnest failed with the error above. 

### Additional information

I did a GIT PULL on kiauh before uninstalling anything so running latest.

tn00364361, 2023-10-11:
I'm having the same problem on Raspberry Pi OS Bookworm on a Pi 4.

ThoHilde, 2024-01-05:
I get the same error. Tried to install crowsnest, but the installation failed and showed as 'Incomplete' in the main menu. Trying to uninstall it results in the same error. Any solutions for this?

jakebullet70, 2024-01-06:
For me I had to wipe the whole OS and start again. Not a Linux guy.

jimmysdiy, 2024-01-30:
seems like an error in the path.

closed 2024-08-31 by dw-0