dw-0/kiauh #388: Creating 8 instance but only can connect 6

dodolcool, 2023-10-03:
### Linux Distribution

Linux mint

### What happened

[image]
[image]
[image]

hello i install 8 klipper instance and moonraker and using mainsail but when adding the printer to mainsail i notice that only 6 instance active and cant connect to the other 2 instance.

### What did you expect to happen

i expected all instance active and can be used.

### How to reproduce

install 8 instance then proceed like usual no special steps required.

how to check if the instance really active or not and is it a way to force activate the instance?

### Additional information

im using PC i3 gen 4 with 8gb ram

dw-0, 2023-10-03:
Since 6 of 8 instances seem to run, i don't think there was an issue setting them up generally.
Instance 6 and 8 are affected it seems, you can try and restart them with the following command:
`sudo systemctl restart klipper-S1ProF` and `sudo systemctl restart klipper-S1ProH`

You can find out if they really startet with
`systemctl status klipper-S1ProF` and `systemctl restart klipper-S1ProH`

Something must be different to the other 6 instances, but as of now i can not tell what it is.

dodolcool, 2023-10-03:
[image]
[image]

since the error port is 7130 and 7132 i tried to check the status of F and H and the result is active.

dw-0, 2023-10-03:
Okay, so Klipper is running fine.
Im sorry i forgot to also give you the commands to check if each Moonraker instance (whats actually matters) is active:
`systemctl status moonraker-S1ProF` and `systemctl status moonraker-S1ProH`

dodolcool, 2023-10-03:
F
[image]
H
[image]

here's the result

dw-0, 2023-10-03:
The second screenshot let me assume that there might be some other process already listening to the port (or ports)?
Try and use this command to see which ports are already in use: `sudo ss -tulpn | grep LISTEN`
Maybe we can see something.

dodolcool, 2023-10-03:
[image]

here's the result

dw-0, 2023-10-03:
Doesn't look like it.
Please upload the klipper and moonraker log of those two instances.

dodolcool, 2023-10-03:
well i cant access those two instance. when add the port to mainsail it will appear with this icon and cant press

[image]

is there any other way to pull the log?

dw-0, 2023-10-03:
Yes there is, you can directly pull those logs from the file system by connecting to it with an SSH client. The logs are located in `~/printer-S1ProF/logs` and `~/printer-S1ProH/logs` then.

dodolcool, 2023-10-03:
[image]
i got this result and it same with H

dw-0, 2023-10-03:
The easiest way might be to use a tool like Filezilla and then pull the logs with it.

dodolcool, 2023-10-03:
[F.zip](https://github.com/dw-0/kiauh/files/12791030/F.zip)
[H.zip](https://github.com/dw-0/kiauh/files/12791052/H.zip)

here's the log i hope it can identify the troble.
also i notice that there's multiple folder with same name like shown below
[image]

dw-0, 2023-10-03:
Look closely, they are not the same names. Compared to Windows, file and foldernames on Linux are case-sensitive. So a folder called "folder" is a different folder than one with the name "Folder".

Maybe that is already the issue here... it seems that you must have installed klipper instances before?
Im not quite sure but maybe the best way would be to start out fresh. Make a backup of all relevant configurations, then remove all instances, remove all those folders, and install all instances fresh.

closed 2023-10-06 by dodolcool