dw-0/kiauh #390: Crowsnest fails to install

BBUK, 2023-10-13:
### Linux Distribution

Fresh Raspbian Lite (32bit) 2023-10-10

### What happened

Compilation of Crowsnest failed with ">>>>>> Something went wrong! Please try again".  

>[100%] Built target datachannel-static
>make[4]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
>make[3]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
>make[2]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
>rm device/buffer_lock.o device/v4l2/buffer.o output/http_hls.o device/camera/camera_output.o device/v4l2/buffer_list.o >device/buffer_queue.o device/buffer.o device/camera/camera_decoder.o device/links.o cmd/list-devices/main.o >device/v4l2/debug.o device/camera/camera_input.o device/v4l2/device_options.o output/http_ffmpeg.o >device/camera/camera.o device/camera/camera_rescaller.o device/libcamera/fake_camera.o cmd/camera-streamer/main.o >device/buffer_list.o device/camera/camera_debug.o util/http/http.o device/camera/camera_pipeline.o >device/dummy/buffer_list.o util/opts/control.o output/http_jpeg.o cmd/camera-streamer/opts.o output/output.o >device/v4l2/device_list.o device/device_list.o device/v4l2/device_media.o device/dummy/device.o device/dummy/dummy.o >util/http/http_methods.o device/dummy/buffer.o cmd/camera-streamer/http.o device/camera/camera_isp.o device/v4l2/v4l2.o >device/device.o device/v4l2/device.o output/http_h264.o
>make[1]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer'
>make: *** [Makefile:62: all] Error 2
>make: *** [Makefile:17: install] Error 2
> >>>>>> Something went wrong! Please try again...

Scrolling up the compilation I see this error:

>util/ffmpeg/remuxer.c: In function ‘ffmpeg_remuxer_open’:
>util/ffmpeg/remuxer.c:97:33: error: initialization discards ‘const’ qualifier from pointer target type [-Werror=discarded->qualifiers]
>   97 |   AVInputFormat *input_format = av_find_input_format(remuxer->input_format);
>      |                                 ^~~~~~~~~~~~~~~~~~~~
>cc1: all warnings being treated as errors
>make[1]: *** [Makefile:92: util/ffmpeg/remuxer.o] Error 1
>make[1]: *** Waiting for unfinished jobs....

Compilation does continue after this point until linking at the end.  Kiauh does, however, report Crowsnest as installed.  Issue looks to be with upstream but seems to have been fixed:  [https://github.com/ayufan/camera-streamer/pull/74](url) and [https://github.com/ayufan/camera-streamer/pull/80](url)

### What did you expect to happen

Crowsnest to install without errors

### How to reproduce

Fresh install of Raspbian Lite, install Klipper, Moonraker, Mainsail and then try to install Crowsnest.

Stephane-Peel, 2023-10-22:
Same Issue here !

Found this thread that explain it : https://github.com/mainsail-crew/crowsnest/issues/188

_"The problem is about camera-streamer not building. Crowsnest is still using an old branch of camera-streamer atm. That branch don't get any updates. Also the newest version of camera-streamer doesn't build atm under bookworm too, as you can read in this issue https://github.com/ayufan/camera-streamer/issues/107.
So the current options we offer would be to stay on bullseye or try to use the legacy branch like described [here](https://crowsnest.mainsail.xyz/faq/use-legacy-branch-on-buster) (not tested under bookworm)."_

Stephane-Peel, 2023-10-22:
solved the issue by manually install camera-streamer "manually" from package : https://github.com/ayufan/camera-streamer/releases/tag/v0.2.7 then restart kiauh.sh install of crowsnest to acheive à working installation

Stephane-Peel, 2023-10-22:
however if it seems like I'm still having issues... 
i'am done...

Switch back to a fresh bullseye install

BBUK, 2023-10-22:
Crowsnest manual install works for me on Bookworm with its legacy/v3 branch.  After cloning the Crowsnest repo switch to legacy/v3 as described here: https://crowsnest.mainsail.xyz/faq/use-legacy-branch-on-buster and then sudo make install.  You should also update moonraker.conf (also as stated on that page).

marklalich, 2023-10-30:
I'm also getting this problem.

I was going to try the route @Stephane-Peel suggested, but when  I run the first command, it says "Unsupported file /home/pi/camera-streamer-generic_#{GIT_VERSION}.bookworm_armhf.deb given in commandline"

What might I be doing wrong?

ovidiogm, 2023-11-03:
I 

> I'm also getting this problem.
> 
> I was going to try the route @Stephane-Peel suggested, but when I run the first command, it says "Unsupported file /home/pi/camera-streamer-generic_#{GIT_VERSION}.bookworm_armhf.deb given in commandline"
> 
> What might I be doing wrong?

I just faced same issue, just download the correct file manually 

This one in my case
[camera-streamer-raspi_0.2.7.bookworm_armhf.deb](https://github.com/ayufan/camera-streamer/releases/download/v0.2.7/camera-streamer-raspi_0.2.7.bookworm_armhf.deb)
then install it with 

sudo apt install "camera-streamer-raspi_0.2.7.bookworm_armhf.deb"

marklalich, 2023-11-03:
Ultimately the issue for me is that KIAUH isn't properly installing Crowsnest new, so the way I got around this was to use another image that has Crowsnest already installed.  Then I SSHed in the my raspberry pi running this new image and installed KIAUH and installed the other packages I want to use.  When you do this and run KIAUH it'll see that Crowsnest is installed and can update it if need be.  You can at this point also install any other packages KIAUH offers and be able to utilize Crowsnest.

tn00364361, 2023-11-07:
I had the same issue but somehow it went away today. I was able to install Crowsnest using KIAUH. Here is my setup:

- Hardware: Raspberry Pi Zero 2W & Pi Camera Module V2.1 (imx219)
- OS: Raspberry Pi OS Bookworm 64bit
- crowsnest: v4.0.5-1-g4050a3bc
- Klipper: v0.11.0-311-g9e765dae
- mainsail: v2.8.0
- mainsail-config: v1.1.0-1-g76a99999
- moonraker: v0.8.0-193-g2e27b073

I'm not sure what has changed but I hope this helps.

Surion79, 2023-11-21:
This is not a KIAUH issue. That issue was about installing crowsnest on Bookworm. It is fixed with the newest crowsnest version.