dw-0/kiauh #393: Repo has untracked source files: ['klippy/extras/gcode_shell_command.py']

Heiko910, 2023-10-18:
### Linux Distribution

Armbian

### What happened

Since I added the GCODE Shell Command Script to my Klipper install I have this Notification next to my klipper version in the Mainsail Update Manager.
Does anyone know what the cause is and what I could try to fix this?

Thanks

[image]

### What did you expect to happen

To have a clean update of Klipper with no error notifications.

### How to reproduce

Good question. I guess it was caused by the Gcode Shell command script

dw-0, 2023-10-18:
Not a kiauh issue or kiauh bug.

closed 2023-10-18 by dw-0

Heiko910, 2023-10-25:
This is what I was told 

[Staubgeborener](https://github.com/Staubgeborener) commented [last week](https://github.com/Staubgeborener/klipper-backup/issues/17#issuecomment-1767664472)
This issue is not related to this project, the [G-Code Shell Command Extension](https://github.com/dw-0/kiauh/blob/master/docs/gcode_shell_command.md) is part of [KIAUH](https://github.com/dw-0/kiauh), so I would suggest opening an issue there or in the official [Mainsail](https://github.com/mainsail-crew) repo.

(But short google search: Seems to be a known issue [here](https://www.reddit.com/r/klippers/comments/158hqxu/repo_has_untracked_source_files/) and https://github.com/mainsail-crew/moonraker-timelapse/issues/128.

dw-0, 2023-10-25:
The Info message about untracked source files you see in mainsail is a fairly recent Mainsail feature. If moonraker reports untracked files, Mainsail now displays them.

As the shell command extension you have installed into Klipper is not an official Klipper file which is tracked in the Klipper repository, Moonraker notices that and reports that.