dw-0/kiauh #394: Menu not refreshed after switching to custom repo

jslaker, 2023-10-28:
### Linux Distribution

Raspberry Pi OS

### What happened

After performing a switch to a custom repo, the user is prompted to "Perform action:" with no indication of what actions are available. The user then has to guess that they need to enter e.g., B for back.

[image]

### What did you expect to happen

KIAUH should display a refreshed menu with available options to the user.

### How to reproduce

Add a custom repo to klipper_repos.txt
- 6 -  Settings
- 1 -  Set up custom Klipper repository
- Select the newly added repo
- Confirm replacing existing install

dw-0, 2023-10-28:
Thanks for reporting. Please update KIAUH and try again. I just pushed a fix for that bug.

closed 2023-10-28 by dw-0

jslaker, 2023-10-28:
Can confirm that I'm now being presented with a menu after repo switch -- thank you!

dw-0, 2023-10-28:
Thanks for confirming!

commit 25dfbb83 (2023-10-28): fix(kiauh): display custom repo dialog again after user interaction

fix #394