dw-0/kiauh #395: KIAUH asks for a sudo password even if sudo does not require a password.

LoganDark, 2023-10-30:
### Linux Distribution

Raspberry Pi OS (Legacy) Lite (Debian Bullseye)

### What happened

KIAUH asks for a sudo password.

### What did you expect to happen

KIAUH should just run sudo, because I have `pi ALL=(ALL) NOPASSWD:ALL` in the sudoers file, and it does not require a password.

### How to reproduce

Put `pi ALL=(ALL) NOPASSWD:ALL` in the sudoers file and then run KIAUH. It will ask for a sudo password despite not needing one.

### Additional information

This makes it impossible to use KIAUH unless you set a password for `pi`, which I don't like to have to do because I prefer key authentication for its security.

EricZimmerman, 2023-12-17:
there is still a password on the account, no? i am in the same boat, NOPASSWD set, public/private key auth, and it worked fine afaik

LoganDark, 2023-12-17:
> there is still a password on the account, no?

But `sudo` does not require one and `sudo` is what this script is using so there's no reason for it to ask for my password.

dw-0, 2023-12-17:
I assume it's a misconfiguration of your sudoers file. I can't reproduce it. Make sure there are no group privileges overwriting the user priviliges.

LoganDark, 2023-12-17:
> I assume it's a misconfiguration of your sudoers file.

It's... not? I can log in at any time and run any command using `sudo`, absolutely no problem. But when I try to run KIAUH, it asks me for a password anyway, even though `sudo` *certainly and absolutely does not require one*. I had to run a `sudo passwd` *just to set a password to use for KIAUH*.

dw-0, 2023-12-17:
Well, as i don't know the content of your sudoers file, i can just make assumptions, simply what i did.

For example, my user is part of the sudo group:
[image]
-> i'l' still get asked for the sudo password.
But:
[image]
-> i'll not be asked for the password.

That's where my assumption came from.

LoganDark, 2023-12-17:
I added my user at the bottom of the file, so that seems fine. I'm no longer using the system that had this issue since I gave up on 3D printing. Can you share what sudo command KIAUH uses in order to ask for the password? Is there a way to just make it not do that?

dw-0, 2023-12-17:
> Can you share what sudo command KIAUH uses in order to ask for the password?

What do you mean what sudo command? There is only one sudo command -> `sudo` 😅
It's mainly used in cases where KIAUH does something in terms of checking system packages. Updating the package list, getting the age of the last update of that list. How should KIAUH prevent a user getting asked for a password it its a system-level thing if not configured otherwise?

LoganDark, 2023-12-17:
Found it:

https://github.com/dw-0/kiauh/blob/7a9e752f9ca3297fbca311d5fbb1d95da3093cac/scripts/ui/install_menu.sh#L38

`sudo -v`: asks the user for a password, no exceptions. Forces sudo to ask for a password even if NOPASSWD is set.

It's not my fault.

dw-0, 2023-12-17:
And `sudo true` won't ask you for password? It does for me.

LoganDark, 2023-12-17:
> And `sudo true` won't ask you for password? It does for me.

That's correct it won't ask me for a password because the sudoers file tells it not to. `sudo -v` forces it to ask for a password even if sudoers says `NOPASSWD:ALL`. `sudo true` only asks for a password if it actually needs one.

dw-0, 2023-12-17:
Okay, i can switch that command to use "true" instead.

closed 2023-12-17 by dw-0

dw-0, 2023-12-17:
Thank you for making me aware of an alternative.

randellhodges, 2024-01-01:
Looks like you missed a spot?

[image]

dw-0, 2024-01-01:
> Looks like it needs adjusted in another spot.
> 
> ![image](https://private-user-images.githubusercontent.com/7727467/293541041-1d3fec33-3d77-483d-a239-3d8bf9a6cad7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxMzUxODcsIm5iZiI6MTcwNDEzNDg4NywicGF0aCI6Ii83NzI3NDY3LzI5MzU0MTA0MS0xZDNmZWMzMy0zZDc3LTQ4M2QtYTIzOS0zZDhiZjlhNmNhZDcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDFUMTg0ODA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2QyZDBjNWMzMGNhNGIwNTY1MDcyMTQ0ZWQwZDYwNmQ5MTEwZmYyOTg0NmNmMzdmMGQzNTA0MjAzN2ZiMDJiNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.aAiZzQQxuihu6S-yHT3XX0UfbfKHpP2DskVuvVPJD_8)

Thank you for the hint! I pushed a commit with that change.

commit f5eb9486 (2023-12-17): refactor(kiauh): use `sudo true` instead of `sudo -v` to prevent password prompts

Fixes #395