dw-0/kiauh #396: KIAUH clears the scrollback buffer every time you use a menu option.

LoganDark, 2023-10-30:
### Linux Distribution

Raspberry Pi OS (Legacy) Lite (Debian Bullseye)

### What happened

I wanted to look at some of the things KIAUH did, but the process of exiting the script caused the entire scrollback buffer to get cleared in my terminal, so now I can't see any history.

### What did you expect to happen

KIAUH should only clear the current screen, not the entire terminal history.

### How to reproduce

Install something, see tons and tons of output, see you can scroll back up. Use the menu to exit the script, see that all the history has disappeared.

closed 2024-08-31 by dw-0

LoganDark, 2024-08-31:
> "Clearing the screen" via `clear` in bash results in the scrollback buffer to be cleared

I know, one can use `clear -x` instead which clears the screen but not the scrollback :) not sure if this is how you fixed it, but if it is fixed, thank you!

dw-0, 2024-08-31:
Well, the "fix" is to not clear anymore at all currently :)

dw-0, 2024-08-31:
I did not know about `clear -x`, this looks actually like a neat way of doing it in the future. Thanks for the hint!

LoganDark, 2024-08-31:
> Well, the "fix" is to not clear anymore at all currently :)

considering the issue was that scrollback was being cleared, simply not clearing the screen would indeed be a fix!

> I did not know about `clear -x`, this looks actually like a neat way of doing it in the future. Thanks for the hint!

❤️