dw-0/kiauh #400: Mainsail Installation Fails

grasshacques, 2023-11-20:
### Linux Distribution

Raspberry Pi Os lite 32 bit (bookworm),
Dietpi (bookworm)

### What happened

This: 
```
 ###### Initializing Mainsail installation ...

###### Create KIAUH backup directory ...
[✓ OK] Directory created!

###### Timestamp: 231119-1713

###### Create backup of /home/math/printer_data/config ...
[✓ OK] Backup created in:
/home/math/kiauh-backups/configs/231119-1713/printer_data

###### Downloading Mainsail from https://github.com/mainsail-crew/mainsail/releases/latest/download/mainsail.zip ...
--2023-11-19 17:13:31--  https://github.com/mainsail-crew/mainsail/releases/latest/download/mainsail.zip
Resolving github.com (github.com)... 64:ff9b::8c52:7004, 140.82.112.4
Connecting to github.com (github.com)|64:ff9b::8c52:7004|:443... failed: Connection timed out.
Connecting to github.com (github.com)|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github.com/mainsail-crew/mainsail/releases/download/v2.8.0/mainsail.zip [following]
--2023-11-19 17:15:44--  https://github.com/mainsail-crew/mainsail/releases/download/v2.8.0/mainsail.zip
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/240875926/3367f3c0-21bf-4f2c-8df7-9baa3da4f07d?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231120%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231120T011544Z&X-Amz-Expires=300&X-Amz-Signature=49ceb7f5ea7a45caf7b0aee84fd5af819c6b77832f7d3b14225305dad6e5a5c7&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=240875926&response-content-disposition=attachment%3B%20filename%3Dmainsail.zip&response-content-type=application%2Foctet-stream [following]
--2023-11-19 17:15:44--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/240875926/3367f3c0-21bf-4f2c-8df7-9baa3da4f07d?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231120%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231120T011544Z&X-Amz-Expires=300&X-Amz-Signature=49ceb7f5ea7a45caf7b0aee84fd5af819c6b77832f7d3b14225305dad6e5a5c7&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=240875926&response-content-disposition=attachment%3B%20filename%3Dmainsail.zip&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 64:ff9b::b9c7:6c85, 64:ff9b::b9c7:6f85, 64:ff9b::b9c7:6e85, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|64:ff9b::b9c7:6c85|:443... failed: Connection timed out.
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|64:ff9b::b9c7:6f85|:443... failed: Connection timed out.
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|64:ff9b::b9c7:6e85|:443... failed: Connection timed out.
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|64:ff9b::b9c7:6d85|:443... failed: Connection timed out.
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 401 Unauthorized

Username/Password Authentication Failed.

#=======================================================#
 Downloading Mainsail from
 https://github.com/mainsail-crew/mainsail/releases/latest/download/mainsail.zip
 failed! 
#=======================================================#

```

### What did you expect to happen

For Mainsail to install and show up as installed in the menu. 

### How to reproduce

New rpi os image preconfigured to use my wifi via pi imager,
Step 1,2, and 3 from kiah git README.md,
install klipper and moonraker via kiauh,
select install mainsail via kiauh

dw-0, 2023-11-22:
Im unsure if this is a KIAUH issue. In the output you provided i see mostly failed connections and a 401 Unauthorized while trying to connect to github. Thats not really something in KIAUHs scope. Did you manage to resolve the issue in the meantime?

closed 2023-12-03 by dw-0

7coil, 2024-04-24:
I've gotten around this problem by disabling IPv6 (not sure why I need to do this)

Add `net.ipv6.conf.all.disable_ipv6 = 1` to `/etc/sysctl.conf`
https://gist.github.com/andreibosco/3badaac477446587bcd6751e186df446