dw-0/kiauh #401: MultiDistribution support

Dedoque, 2023-11-20:
### Is your feature request related to a problem? Please describe

With os like arch linux installation is impossible becouse apt hardcoded in unreadable scripts.

### Describe the solution you'd like

I suggest to use for install system dependencies some one of scm, like [Ansible](https://docs.ansible.com/ansible/latest/getting_started/index.html) or  [Salt-ssh](https://docs.saltproject.io/salt/user-guide/en/latest/topics/salt-ssh.html).
Described onece -- for many distribs -- Ded/rhel/arch and bsd systems. Simply conditions for different packages and many other features has suppor from box!
No needs difficult bash scripts -- Unreadable, difficult to maintain -- Blood from the eyes!

dw-0, 2023-11-22:
Thanks for the request. A feature like this would likely mean, that there is more maintenance involved with dependencies. Currently dependencies are simply read from the official install script of klipper for example. Thats written with debian based systems in mind. But as im currently working on a rewrite of KIAUH in python, we could think about a different, a new approach of handling each components dependencies to allow other distros. This would also result in much more testing though. I have added the feature request to the "next" milestone. For now it will not be part of the v6 release.

Dedoque, 2023-11-25:
>> is more maintenance involved with dependencies.
Based deps: Python and git at all!
Scm just a method of maintaince configuration and state of system.

>> Thats written with debian based systems in mind. But as im currently working on a rewrite of KIAUH in python, we could think about a different, a new approach of handling each components dependencies to allow other distros.
Same, but little-a-bit simply, maybe. Look at [poetry](https://python-poetry.org/docs/) depends and script manager,
and [click](https://click.palletsprojects.com/en/8.1.x/) cmd interfacer.

Do not make a new bicicle, maybe some one already make it))

dw-0, 2023-11-25:
I want to stick to the python standard library and don't want KIAUH to have special python dependencies.
The most complicated part of the TUI is already rewritten :)

Dedoque, 2023-11-30:
>>don't want  KIAUH to have special python dependencies.
<< that's right!
And there is no need for this! 
Keep it up!
We wait!

BiloxiGeek, 2024-01-17:
Is there any sort of timeline on the python V6 release?  

I started looking deeper into kiauh to see if I can get it working on an rpm based system, specifically Alma Linux but that would mean the popular rpm based distros would likely work as well.  I'd rather not spend the time digging that deep into kiauh if the rewrite is in the works and coming soon.

I got a fairly high amount of experience with RHEL style systems and I'm willing to help test the rewrite.

dw-0, 2024-01-17:
Hi @BiloxiGeek ,
currently there is no ETA for a possible release. The current development still focuses on debian based systems as well, i have not added any generic "use package manager x now" functions implemented. The current state is, that the regular "klipper stack" is working. That means installation, updating and removing of Klipper, Moonraker and currently only Mainsail. Fluidd will follow next probably. All other features are still in the pipeline. So i would not use the term "soon" at the moment, im sorry.

Mr-Boshi, 2024-06-18:
How about stating in the Prerequisites section of README that apt is a must then?

lexfrei, 2025-03-16:
I'm working on it.
I really hate deb-based distros, and I really want to migrate to Rocky on my printer too.
It's not really hard to make the tools work, but it will take a lot to find all packages alternatives for other distros and test them. I need help with arch and suse testing (or any other zypper and pacman systems).
Not just the package managers should be supported, but system managers too.
Currently, I'm aiming to support Rocky and Alpine first and other distros secondary.
Also, it would be great to add tests for this. I need to test utils agains some distros, may I create CI part for this? @dw-0 what do you think about this kind of testing in the project?

dw-0, 2025-03-19:
> It's not really hard to make the tools work, but it will take a lot to find all packages alternatives for other distros and test them. 

And this is the worst part and can become a huge maintenance burden on KIAUHs side.
Usually i prefer each component to maintain their required dependencies, which then can be parsed by KIAUH.
A perfect example for that is Moonraker. The maintainer of the project already laid a foundation for future support for distros other than debian based ones. 

> what do you think about this kind of testing in the project?

I think the project has reached a point where sooner or later there will be no way around testing. A CI part would be a logical addition. Feel free to propose an initial approach of some first simple unit tests (PyTest?) maybe and a corresponding CI workflow.