dw-0/kiauh #403: Please explain why 64 bit Raspberry Pi OS is not recommended

RichardBronosky, 2023-11-22:
### Is your feature request related to a problem? Please describe

As recently as 2023/03/08 the `README` recommends using the 32 bit OS.

https://github.com/dw-0/kiauh/blame/6fcd7a3f0854588667596c8c1f44a211b10ca60d/README.md#L32

Please explain why this is. It would be valuable to at least know what is lost by choosing the 64 bit Raspberry Pi OS.

### Describe the solution you'd like

1. Describe what is lost by choosing the 64 bit Raspberry Pi OS.
2. Describe what is lost by choosing the 32 bit Raspberry Pi OS.

dw-0, 2023-11-22:
I recommend this as an example to pick, just because it's likely the most tested and used variant out there. Personally, i have no hardware which would benefit from a 64bit OS, so personally i see no point in choosing it over the 32bit. Apart from that, i was told that there were no real issues anymore. So you could also pick 64bit if that's what you desire.

RichardBronosky, 2023-11-22:
Searching for "64 bit" on https://discord.com/channels/431557959978450984/601514366256283667/1173766937084448839 turns up a lot of results.

If I were to engage with the community to aggregate the facts an opinions and label each as such, would you accept a pull request?

I began my career in 1995 thanks to Open Source software. I am always to looking for ways to give back to the community that saved me from the Appalachian coal mines that I was born above.

dw-0, 2023-11-22:
> If I were to engage with the community to aggregate the facts an opinions and label each as such, would you accept a pull request?

Sure, feel free to evaluate if 64bit is absolutely fine to use (i assume it is) and we can either simply remove the "bit recommendation" or add both to it. The whole instructions thing is just for the sake of having SOME instructions to people who might feel completely lost otherwise and have no idea how to begin in the first place. So i just put together what seemed to work for the most and was usually recommended within the community.

EricZimmerman, 2023-12-17:
64 bit bookworm here on 2 printers, used kiauh on both, zero issues

dw-0, 2023-12-17:
> 64 bit bookworm here on 2 printers, used kiauh on both, zero issues

Thanks for the reply and feedback.

commit 5ebe9411 (2023-12-17): readme: don't recommend a specific architecture anymore

Fixes #403 

Both 32bit and 64bit images were reported to work fine. So we don't need a strict 32bit recommendation anymore which may confuses users, why they should pick 32bit over 64bit.
closed 2023-12-17 by dw-0