dw-0/kiauh #404: Broken envs after distro upgrades

freman, 2023-11-22:
### Linux Distribution

Was raspbian 10, then 11, now 12 (I was upgrading so I could get webcams working)

### What happened

First sign something is wrong is when you l aunch kiauh and it says the following as it's starting up (tho for some reason I didn't notice it)

```
/home/pi/kiauh/scripts/klipper.sh: line 557: ((: not
not
setting
3: syntax error in expression (error token is "not
setting
3")
```

When you go ahead and update klipper you get this error

```
###### Reading dependencies...
/home/pi/kiauh/scripts/klipper.sh: line 310: ((: not
not
setting
3: syntax error in expression (error token is "not
setting
3")
/home/pi/kiauh/scripts/klipper.sh: line 313: ((: not
not
setting
3: syntax error in expression (error token is "not
setting
3")
>>>>>> Internal Error: missing parameter 'python_version' during function call of install_klipper_packages
```

I did some googling and trawling through source (and sourcing the source and using `set -x` to debug) and then poked python

```
"${KLIPPY_ENV}"/bin/python --version 2>&1 | cut -d" " -f2 | cut -d"." -f1
not
not
setting
3
```

This results in the version being pulled but not updated, when you run kiauh again it says it's the latest version but nothing actually works (nginx returns an error)

Moonraker suffers the same way

### What did you expect to happen

Installing updates after updating distro

### How to reproduce

Have a working install of everything on raspbian 10, then upgrade to raspbian 11 - it's gonna be because python version changes

### Additional information

Resoution for me looked something like

```bash
rm -fr moonraker-env/
virtualenv moonraker-env
./moonraker-env/bin/pip install -r moonraker/scripts/moonraker-requirements.txt
```

Then using kiauh to roll them back to a previous version, then run the update again.

Mostly writing to document this issue because that took me an unusually long time to debug.

dw-0, 2023-11-22:
As KIAUH won't do distro updates, and they were done by you manually, it can't be considered as a KIAUH bug.
Apart from that, i got the confirmation that KIAUH works with Debian Buster, Bullseye and Bookworm.
Distro upgrades can have sideeffects and things break. Seems like that was the case here.

freman, 2023-11-22:
Oh yeh I'm perfectly aware I'm working well outside the scope of anything KIAUH would normally see (and I'd be a heck of an edge case) hence, I'm happy with it to live in the ether forever as documentation :D

dw-0, 2023-11-22:
That's fine with me :) But I will still close the issue then :D

closed 2023-11-22 by dw-0