dw-0/kiauh #405: KIAUH Installation of Crowsnest failing on Bookworm

awiltschek, 2023-11-26:
### Linux Distribution

Raspi OS Lite 32Bit Bookworm

### What happened

device/libcamera/buffer_list.cc: In function ‘int libcamera_buffer_list_open(buffer_list_t*)’:
device/libcamera/buffer_list.cc:103:21: error: ‘using element_type = class libcamera::CameraConfiguration’ {aka ‘class libcamera::CameraConfiguration’} has no member named ‘transform’
  103 |     configurations->transform |= libcamera::Transform::VFlip;
      |                     ^~~~~~~~~
device/libcamera/buffer_list.cc:106:21: error: ‘using element_type = class libcamera::CameraConfiguration’ {aka ‘class libcamera::CameraConfiguration’} has no member named ‘transform’
  106 |     configurations->transform |= libcamera::Transform::HFlip;
      |                     ^~~~~~~~~
device/libcamera/buffer_list.cc:108:26: error: ‘using element_type = class libcamera::CameraConfiguration’ {aka ‘class libcamera::CameraConfiguration’} has no member named ‘transform’
  108 |   if (!!(configurations->transform & libcamera::Transform::Transpose)) {
      |                          ^~~~~~~~~
make[1]: *** [Makefile:99: device/libcamera/buffer_list.o] Error 1
rm device/buffer_lock.o device/v4l2/buffer.o output/http_hls.o device/camera/camera_output.o device/v4l2/buffer_list.o device/buffer_queue.o util/ffmpeg/remuxer.o device/buffer.o device/camera/camera_decoder.o device/links.o cmd/list-devices/main.o device/v4l2/debug.o device/camera/camera_input.o util/opts/opts.o device/v4l2/device_options.o output/http_ffmpeg.o device/camera/camera.o device/camera/camera_rescaller.o device/libcamera/fake_camera.o cmd/camera-streamer/main.o device/buffer_list.o device/camera/camera_debug.o util/http/http.o device/camera/camera_pipeline.o device/dummy/buffer_list.o util/opts/control.o output/http_jpeg.o cmd/camera-streamer/opts.o util/opts/fourcc.o output/output.o device/v4l2/device_list.o device/device_list.o device/v4l2/device_media.o device/dummy/device.o device/dummy/dummy.o util/http/http_methods.o cmd/camera-streamer/status.o device/dummy/buffer.o cmd/camera-streamer/http.o util/opts/log.o device/camera/camera_isp.o device/v4l2/v4l2.o device/device.o device/v4l2/device.o device/libcamera/buffer.o output/http_h264.o
make[1]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer'
make: *** [Makefile:66: all] Error 2
make: *** [Makefile:19: install] Error 2
>>>>>> Something went wrong! Please try again...

### What did you expect to happen

that it just works :-)

### How to reproduce

Clean Image of RASPI OS Lite Bookworm 32 Bit, Install Klipper, Install Moonraker, Install Mainsail, Install Crowsnest - bang

roykrikke, 2023-11-27:
I have the same issue:
I have the same issue:
```shell
[ 90%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/pollinterrupter.cpp.o
[ 92%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/pollservice.cpp.o
[ 92%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/tcpserver.cpp.o
[ 92%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/tcptransport.cpp.o
[ 95%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/tlstransport.cpp.o
[ 95%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/verifiedtlstransport.cpp.o
[ 95%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/websocket.cpp.o
/home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp: In constructor ‘rtc::impl::TlsTransport::TlsTransport(std::shared_ptr<rtc::impl::TcpTransport>, std::optional<std::__cxx11::basic_string<char> >, rtc::impl::certificate_ptr, rtc::impl::Transport::state_callback)’:
/home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp:351:58: warning: ‘void EC_KEY_free(EC_KEY*)’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  351 |                 auto ecdh = unique_ptr<EC_KEY, decltype(&EC_KEY_free)>(
      |                                                          ^~~~~~~~~~~
In file included from /usr/include/openssl/x509.h:33,
                 from /usr/include/openssl/ssl.h:31,
                 from /home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tls.hpp:56,
                 from /home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/certificate.hpp:24,
                 from /home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.hpp:22,
                 from /home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp:19:
/usr/include/openssl/ec.h:1003:28: note: declared here
 1003 | OSSL_DEPRECATEDIN_3_0 void EC_KEY_free(EC_KEY *key);
      |                            ^~~~~~~~~~~
/home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp:352:45: warning: ‘EC_KEY* EC_KEY_new_by_curve_name(int)’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  352 |                     EC_KEY_new_by_curve_name(NID_X9_62_prime256v1), EC_KEY_free);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/usr/include/openssl/ec.h:998:31: note: declared here
  998 | OSSL_DEPRECATEDIN_3_0 EC_KEY *EC_KEY_new_by_curve_name(int nid);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~
/home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp:352:80: warning: ‘void EC_KEY_free(EC_KEY*)’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  352 |                     EC_KEY_new_by_curve_name(NID_X9_62_prime256v1), EC_KEY_free);
      |                                                                                ^
/usr/include/openssl/ec.h:1003:28: note: declared here
 1003 | OSSL_DEPRECATEDIN_3_0 void EC_KEY_free(EC_KEY *key);
      |                            ^~~~~~~~~~~
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/websocketserver.cpp.o
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/wstransport.cpp.o
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/wshandshake.cpp.o
[100%] Linking CXX static library libdatachannel-static.a
make[5]: Leaving directory '/home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
[100%] Built target datachannel-static
make[4]: Leaving directory '/home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[3]: Leaving directory '/home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[2]: Leaving directory '/home/user/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
rm device/buffer_lock.o device/v4l2/buffer.o output/http_hls.o device/camera/camera_output.o device/v4l2/buffer_list.o device/buffer_queue.o util/ffmpeg/remuxer.o device/buffer.o device/camera/camera_decoder.o device/links.o cmd/list-devices/main.o device/v4l2/debug.o device/camera/camera_input.o util/opts/opts.o device/v4l2/device_options.o output/http_ffmpeg.o device/camera/camera.o device/camera/camera_rescaller.o device/libcamera/fake_camera.o cmd/camera-streamer/main.o device/buffer_list.o device/camera/camera_debug.o util/http/http.o device/camera/camera_pipeline.o device/dummy/buffer_list.o util/opts/control.o output/http_jpeg.o cmd/camera-streamer/opts.o util/opts/fourcc.o output/output.o device/v4l2/device_list.o device/device_list.o device/v4l2/device_media.o device/dummy/device.o device/dummy/dummy.o util/http/http_methods.o cmd/camera-streamer/status.o device/dummy/buffer.o cmd/camera-streamer/http.o util/opts/log.o device/camera/camera_isp.o device/v4l2/v4l2.o device/device.o device/v4l2/device.o device/libcamera/buffer.o output/http_h264.o
make[1]: Leaving directory '/home/user/crowsnest/bin/camera-streamer'
make: *** [Makefile:66: all] Error 2
make: *** [Makefile:19: install] Error 2
>>>>>> Something went wrong! Please try again...
```

ashleyjkell, 2023-11-27:
Looks like an issue in camera streamer:

https://github.com/ayufan/camera-streamer/issues/118