dw-0/kiauh #408: Support for Octoapp

pauluchin, 2023-12-01:
### Is your feature request related to a problem? Please describe

Last ver support for klipper  and octoapp companion 

can you add in the script?

bash -c "$(curl -fsSL https://octoapp.eu/install.sh)"

Thants

### Describe the solution you'd like

.

### Describe alternatives you've considered

.

dw-0, 2023-12-01:
Thanks for your request.
Im currently not interested in adding that install script to KIAUH. I don't think i will be either in the future.
In the next KIAUH version i plan to drop support for everything regarding OctoPrint, hence adding this script doesn't make much sense currently. I will keep this FR open, until a final decision was made. Thanks again!

jimmysdiy, 2024-01-30:
@dw-0 OctoApp is a Companion App that is for OctoPrint and Klipper. While OctoPrint doesn’t natively support Klipper. I think OctoApp is making every effort to work with Mainsail & Fluidd. I am consistently in touch with the OctoApp Dev. I think having this script to install will provide broader opportunities

dw-0, 2024-03-02:
It might become a community extension in the future if there is real interest getting it added to KIAUH. Though, i can't guarantee that i will implement it myself (likely not) and so it will require another contributor to do so.

closed 2024-04-14 by dw-0