dw-0/kiauh #413: RPC failed;

DiscusZ, 2023-12-13:
### Linux Distribution

raspberrypi debian 10 and also 11

### What happened

trying to install klipper and thefollowing

Cloning into '/home/pi/klipper'...
remote: Enumerating objects: 55013, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (17/17), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 1711 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: index-pack failed

### What did you expect to happen

it should not error like this

### How to reproduce

every time I try and install klipper it happens

dw-0, 2023-12-13:
Looks like a network issue. Nothing in KIAUHs scope.

JonasV328, 2023-12-20:
I had a similar problem when installing Crowsnest with KIAUH.

It seems that
`git config --global http.postBuffer 157286400`
before running KIAUH fixed the problem for me.

DiscusZ, 2023-12-20:
I will try that larger value sometime.  I did tinker with some values and
thruball failed.  I ended up reinstalling from scratch

closed 2023-12-25 by dw-0