dw-0/kiauh #414: Unattentended / noninteractive examples?

jedahan, 2023-12-20:
### Is your feature request related to a problem? Please describe

I'd like to setup CI to build images for Raspberry Pi OS + klipper/moonraker/fluidd. Having a way to script kiauh would help greatly.

### Describe the solution you'd like

Same solution as raspi_config - expose functions as cli subcommands. Shouldn't need the `nonint` prefix.

dw-0, 2023-12-25:
Thanks for the feature request. That is probably something that is not planned in the near future (if at all). I keep this request open anyways, for future tracking.

HumbleDeer, 2024-01-23:
Scripting KIAUH would be somewhat cumbersome when KIAUH itself is just a script. You could call the individual sub-modules and use the stdin fill-in methods you'd use in other scripts to interact with it.

closed 2024-03-02 by dw-0