dw-0/kiauh #415: could not build wheels for pillow

jonawald, 2023-12-23:
### Linux Distribution

Debian Buster running on Moto G7 Play phone

### What happened

         self.build_extensions()
        File "/tmp/pip-install-pzanuu1c/pillow_b2a86a11f3534e43a8840ee01ac8f7a1/setup.py", line 809, in build_extensions
          raise RequiredDependencyException(f)
      __main__.RequiredDependencyException: zlib
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "<string>", line 36, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-pzanuu1c/pillow_b2a86a11f3534e43a8840ee01ac8f7a1/setup.py", line 1010, in <module>
          raise RequiredDependencyException(msg)
      __main__.RequiredDependencyException:
      
      The headers or library files could not be found for zlib,
      a required dependency when compiling Pillow from source.
      
      Please see the install instructions at:
         https://pillow.readthedocs.io/en/latest/installation.html
      
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pillow
  Running setup.py clean for pillow
Failed to build pillow
ERROR: Could not build wheels for pillow, which is required to install pyproject.toml-based projects

### What did you expect to happen

Moonraker did not install failed with the above error. I have done this several times. It always consistently fails at this junction. 

### How to reproduce

Install Klipper on Moto G7 Play phone using Debian Buster. Follow the instructions [here ](https://github.com/gaifeng8864/klipper-on-android/blob/main/README_en.md). When using Kiauh to install moonraker I get this error.

dw-0, 2023-12-25:
It's likely not an issue with KIAUH. If you have any issues when following third party instructions, please contact the creator of those instructions for support. KIAUH is tested and reported to work on Debian Buster, though, that does not include Debian Buster installs running in Android Smartphones.

closed 2023-12-25 by dw-0

mactep, 2024-07-18:
For the sake of documentation, the problem seems to be caused by non default installation of those libraries. In my case I managed to fix it by finding where they were installed and then linking those files to the "standard" library path.

- Use `find / -name libz.so 2>/dev/null` to find where the libraries were installed. The `2>/dev/null` is there to filter the "permission denied" errors. In my case they were located in `/usr/lib/arm-linux-gnueabihf/`.

- Place the libraries somewhere they are being searched, like so: `sudo ln -s /usr/lib/arm-linux-gnueabihf/libz.* /usr/lib/`.

Now Pillow should build without issues, but in case you find additional errors, you can try some things listed in the official docs, like installing the dependencies and setting `CFLAGS` to the directory your `libz.so` is currently at. After this you'll most likely going to have the same error for `libjpeg.so`, so just do the same for it.